CroatianMeteorNetwork/RMS #7: [Feature request] more analyze on meteor

trlsmax, 2018-11-07:
Hi
Can we do more analyze on the meteor after detected ? Just like [UFOAnalyzer
](http://sonotaco.com/soft/e_index.html#ufoa) do. For example, which group is the meteor from , multi site data analyze for orbit of the meteor.

Or, just generate the files which UFOAnalyzer need and analyze them in UFOAnalyzer.

dvida, 2018-11-07:
Hi Max,
there there are tools to do the astrometric and the photometric calibration, please read this manual: https://github.com/CroatianMeteorNetwork/RMS/blob/master/Guides/SkyFit.md

It is not completely up to date because I keep adding features every day, but you can easily make an astrometric plate using this guide.
If you want to manually reduce an event and get its RA, Dec, magnitude, you can use Utils.ManualReduction. I still don't have a guide for it, but the features are fairly well explained in the program itself. You can use RMS.Astrometry.ApplyAstrometry to apply the calibration (platepar file) to an FTPdetectinfo file.

Now, regarding trajectory estimation and meteor shower association, we still don't have those tools at our disposal. The idea is that the observations are sent to the server and that everything is done automatically on the server. I have the trajectory code, but it is still not public - it will be released once I publish a paper about it, which will happen in a few months. I expect to have the basic serverside trajectory estimation working within a year.

On the other hand, I guess we could parse the RMS format to UFO format, I don't see why not. Could you please send me examples of UFO input formats so I can write a format converter?

trlsmax, 2018-11-08:
Thank you for reply. I will prepare some samples for you. But analyze it all in server automaticly will be great. Looking forward to it.

dvida, 2018-11-23:
RMS2UFO is now one of the tools in Utils that can be used for format conversion, and every night after recording the FTPdetectinfo will be converted into the UFOOrbit input CSV.

closed 2018-11-23 by dvida