CroatianMeteorNetwork/RMS #11: Minimum FPS?

dbainbridge, 2019-01-07:
What is the minimum FPS that the software will still work at detecting meteors?  Is it required to have multiple images with the same meteor in it or can it detect a meteor in a single long exposure?

mrprompt, 2019-01-07:
The FPS is restrict from camera/input device, 25.000 or 29.000 FPS is the default for NTFS and PAL systems, if you change this, maybe the camera doesn't work correctly.

dvida, 2019-01-08:
Hi,
so the detector will not work on static images of meteors. It needs videos and the meteors need to last at least 4 frames to be detected. The 4 frame limits is basically the limiting factor, most meteors last around 0.5 seconds, so basically anything less then 10 FPS will not work. A minimum of 25 FPS is recommended.
But, if you have a static image of a meteor, it can still be manually reduced using Utils.ManualReduction (you can either pick the shutter breaks or the beginning and ending point of the meteor). You can generate the astrometric and photometric calibration using RMS.Astrometry.SkyFit and use it during manual reduction to get RA and Dec of the selected points.

closed 2019-01-08 by dvida