CroatianMeteorNetwork/RMS #12: What camera hardware is supported?

dbainbridge, 2019-01-07:
Could documentation somewhere explain how the software is connecting to a camera and what someone needs to watch for when buying a camera to make sure it is compatible with the software?  I only noticed the ReadMe mention recommended sony sensors to use but it was not clear what "driver" is being used to talk to the camera.  I assume it is something to do with the OpenCV gstreamer but further details explaining how this works would be helpful.

mrprompt, 2019-01-07:
The SONY camera is recommended because is the camera used in tests. You can use any camera with a good capture in night sky (like 0.001 LUX).

dvida, 2019-01-08:
I have added a section about this in the readme. So basically any IP cameras will work, or any video capture devices that can be installed on Linux (e.g. EasyCap or any device that works on v4l2). I can't really provide any concrete pointers, but try to get something with the IMX291 or IMX307 sensor. Also, please take a look at our [website](http://globalmeteornetwork.org/) for more info, and join the groups.io mailing list.

closed 2019-01-08 by dvida

Cybis320, 2024-01-09:
Not sure about the rpi camera module low light performance. 
FYI, I was able to purchase the IMX291 recently without issues from AliExpress. The low light performance is good at that price point. 

Here's the one I last ordered:
https://a.aliexpress.com/_mNPZ3K0

F0.95 Aperture M16 Mount with IR switch (great for night and day):
https://a.aliexpress.com/_mrWRiO8

Maybe the buying guide needs updating.

bcasari, 2024-01-10:
@Cybis320 
Many thanks for your answer!

I have looked around on the internet and it appears everyone recommends the same camera module IMX291. Do you know why no one attempts to use the Raspberry Pi Camera Modules (with IMX219 NoIR) in Meteor Cameras? Is there a reason I am overlooking or is no one attempting to use them because no one recommends them? 

The advantage of Raspberry Pi Cameras is that they are in a similar price range as the IMX291, have good support with Raspberry Pi and are easier to find.

dvida, 2024-01-10:
Thanks for the question - in fact, people have tried them but they are simply not sensitive enough to capture enough light at video frame rates. They're ok for long exposures, but that is not sufficient for fainter meteors.
Alternatives similar to the RPi camera module exist, such as the MIPI variants of the IMX291 and the IMX307. But the IP version of the 291 is the simplest and most convenient to use, and also has the best video quality.

bcasari, 2024-01-10:
@dvida 
Thank you! It makes sense to stick to those cameras then.

adalava, 2024-01-10:
Just to add, back in 2017/2018 I've replaced original F2.0 RPi NoIR lenses by an alternative F0.95, and by a CS mount F1.0 lenses. Even with the modification it was still not sensitive enough.