CroatianMeteorNetwork/RMS #16: gstreamer does not really work on Stretch...

apevec, 2019-05-06:
"..., and it's necessary if you want to run an IP camera"

^ quote from README. Is there more details, Debian or upstream gstreamer bug report to track?

dvida, 2019-05-06:
This is the thread with the latest developments: https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=200306&start=50#p1456770
The latest post is from a few days ago so the issue might be fixed. It would be great if someone would try it and report back.

dvida, 2021-04-25:
We fortunately found a workaround and it works well when this device string is used:
device: rtspsrc protocols=tcp tcp-timeout=5000000 retry=5 location="rtsp://192.168.42.10:554/user=admin&password=&channel=1&stream=0.sdp" latency=1000 ! rtpjitterbuffer ! rtph264depay ! h264parse ! v4l2h264dec ! appsink max-buffers=25 drop=true sync=1

closed 2021-04-25 by dvida