CroatianMeteorNetwork/RMS #21: Specify latest py2 compatible astropy

apevec, 2019-10-18:
Reported by Maciej in https://groups.io/g/globalmeteornetwork/message/468

Processing dependencies for RMS==0.1
Searching for astropy>=2.0.3
Reading https://pypi.org/simple/astropy/
Downloading https://files.pythonhosted.org/packages/ea/63/02ada19599c7ad34b4984e797c555cacf2c358d48824ffe331e029574a2c/astropy-3.2.2.tar.gz#sha256=8553f4a95e9938a0c7ae520633f7172356824b8eb324dd967fca6baf00ac19bf
Best match: astropy 3.2.2
Processing astropy-3.2.2.tar.gz
Writing /tmp/easy_install-khgwDY/astropy-3.2.2/setup.cfg
Running astropy-3.2.2/setup.py -q bdist_egg --dist-dir /tmp/easy_install-khgwDY/astropy-3.2.2/egg-dist-tmp-YUOEAh
ERROR: Python >=3.5 is required by astropy

apevec, 2019-10-18:
Collecting astropy~=2.0.3
  Downloading https://files.pythonhosted.org/packages/bc/ad/f108a5bd8aba3d5f5d4d5090a8802a78e4e50e8150cb090b6d43b399db5f/astropy-2.0.15-cp27-cp27mu-manylinux1_x86_64.whl (6.1MB)

apevec, 2019-10-18:
TODO make it work with both py2 and py3

mrprompt, 2020-09-21:
@apevec you can add these two libs into requirements.txt file, it's like:

```
astropy~=3.2.2; python_version <= '3.7'
astropy~=2.0.3; python_version >= '2.7'
```

apevec, 2024-03-26:
not needed, originally reported issue is NOT reproducible
astropy>=2.0.3 in requirements.txt works fine, installing latest compatible version for the Python2 https://pypi.org/project/astropy/2.0.16/

closed 2024-03-26 by apevec