CroatianMeteorNetwork/RMS #28: platepar

WirthmU, 2020-05-21:
I got the following issues when running RMS:

`2020/05/21 15:58:28-INFO-DownloadPlatepar-line:23 - Checking for new platepar on the server...
2020/05/21 15:58:28-DEBUG-DownloadPlatepar-line:24 - Establishing SSH connection to: gmn.uwo.ca:22...
2020/05/21 15:58:29-ERROR-UploadManager-line:51 - Failed loading /root/.ssh/id_rsa[Errno 2] No such file or directory: u'/root/.ssh/id_rsa'
2020/05/21 15:58:29-INFO-Reprocess-line:80 - No platepar file found!

Any help to this?

dvida, 2020-05-21:
Hi!
It's not really an error, but just a warning telling you that the station is not able to connect to the server. Once you install the camera, send me the public SSH key and I'll give you a station code if you'd like to participate in the network.

WirthmU, 2020-09-19:
> Antworten an: CroatianMeteorNetwork/RMS <reply@reply.github.com>

Hello Denis,

 

I finally got my camera up and running on my home LAN. 

 

For I do not know what reason I got a camera error:

 

This is the camera setting in the .config file:

device: rtspsrc location=rtsp://192.168.1.59:554/user=admin&password=&channel=1&stream=0.sdp ! rtph264depay ! h264parse ! decodebin ! videoconvert ! appsink sync=1

 

 

If I try to do preview in gstreamer:

gst-launch-1.0 rtspsrc location="rtsp://192.168.1.59:554/user=admin&password=&channel=1&stream=0.sdp" ! rtph264depay ! h264parse ! decodebin ! autovideosink

 

I get the following:

Setting pipeline to PAUSED ...

Pipeline is live and does not need PREROLL ...

Got context from element 'autovideosink0': gst.gl.GLDisplay=context, gst.gl.GLDisplay=(GstGLDisplay)"\(GstGLDisplayX11\)\ gldisplayx11-0";

Progress: (open) Opening Stream

Progress: (connect) Connecting to rtsp://192.168.1.59:554/user=admin&password=&channel=1&stream=0.sdp

Progress: (open) Retrieving server options

Progress: (open) Retrieving media info

Progress: (request) SETUP stream 0

Progress: (open) Opened Stream

Setting pipeline to PLAYING ...

New clock: GstSystemClock

Progress: (request) Sending PLAY request

Progress: (request) Sending PLAY request

Progress: (request) Sent PLAY request

WARNING: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: Delayed linking failed.

Additional debug info:

./grammar.y(510): gst_parse_no_more_pads (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:

failed delayed linking some pad of GstRTSPSrc named rtspsrc0 to some pad of GstRtpH264Depay named rtph264depay0

ERROR: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstUDPSrc:udpsrc0: Internal data stream error.

Additional debug info:

gstbasesrc.c(3055): gst_base_src_loop (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstUDPSrc:udpsrc0:

streaming stopped, reason not-linked (-1)

Execution ended after 0:00:02.150704152

Setting pipeline to PAUSED ...

Setting pipeline to READY ...

Setting pipeline to NULL ...

Freeing pipeline ...

 

I am using the camera you suggested.

 

Any help to this?

 

Thank you

 

Urs

mrprompt, 2020-09-20:
@WirthmU  it's not a bug. You can create a new key using `ssh-keygen` into terminal and send the `.pub` file created into `.ssh` folder in your home dir (ex.: /home/pi/.ssh).

WirthmU, 2020-09-20:
Seems to work now - except for uploading data.
I get the following error:
```
2020/09/20 22:21:37-ERROR-UploadManager-line:183 - Raw
Traceback (most recent call last):
  File "/home/uw/vRMS/RMS/UploadManager.py", line 111, in uploadSFTP
    t.start_client()
  File "/usr/local/lib/python3.6/dist-packages/paramiko-2.7.1-py3.6.egg/paramiko/transport.py", line 660, in start_client
    raise e
  File "/usr/local/lib/python3.6/dist-packages/paramiko-2.7.1-py3.6.egg/paramiko/transport.py", line 2083, in run
    self._handler_table[ptype](self, m)
  File "/usr/local/lib/python3.6/dist-packages/paramiko-2.7.1-py3.6.egg/paramiko/transport.py", line 2199, in _negotiate_keys
    self.kex_engine.start_kex()
  File "/usr/local/lib/python3.6/dist-packages/paramiko-2.7.1-py3.6.egg/paramiko/kex_curve25519.py", line 54, in start_kex
    serialization.Encoding.Raw, serialization.PublicFormat.Raw
  File "/usr/lib/python3.6/enum.py", line 326, in __getattr__
    raise AttributeError(name) from None
AttributeError: Raw
2020/09/20 22:21:37-WARNING-UploadManager-line:371 - Uploading failed! Retry 5 of 5
2020/09/20 22:21:39-INFO-StartCapture-line:664 - Closing upload manager...
```

dvida, 2020-09-21:
Hi,
this happens because the ssh-public key you generated is not on the server. What station ID are you using?
Could you send me the public ssh key that is in ~/.ssh/id_rsa.pub
You can get the contents of that file by running:
cat ~/.ssh/id_rsa.pub

WirthmU, 2020-09-24:
It is working now - ty!

closed 2020-09-24 by WirthmU