CroatianMeteorNetwork/RMS #37: Python 3.8

jenyak, 2020-09-27:
I got this error during installation on python 3.8.5
```
$ sudo python3 setup.py install
running install
running bdist_egg
running egg_info
writing RMS.egg-info/PKG-INFO
writing dependency_links to RMS.egg-info/dependency_links.txt
writing requirements to RMS.egg-info/requires.txt
writing top-level names to RMS.egg-info/top_level.txt
reading manifest file 'RMS.egg-info/SOURCES.txt'
writing manifest file 'RMS.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
running build_ext
building 'RMS.Astrometry.CyFunctions' extension
x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/lib/python3/dist-packages/numpy/core/include -I/usr/lib/python3/dist-packages/numpy/core/include -I/usr/include/python3.8 -c RMS/Astrometry/CyFunctions.c -o build/temp.linux-x86_64-3.8/RMS/Astrometry/CyFunctions.o
In file included from /usr/lib/python3/dist-packages/numpy/core/include/numpy/ndarraytypes.h:1822,
                 from /usr/lib/python3/dist-packages/numpy/core/include/numpy/ndarrayobject.h:12,
                 from /usr/lib/python3/dist-packages/numpy/core/include/numpy/arrayobject.h:4,
                 from RMS/Astrometry/CyFunctions.c:614:
/usr/lib/python3/dist-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:17:2: warning: #warning "Using deprecated NumPy API, disable it with " "#define NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
   17 | #warning "Using deprecated NumPy API, disable it with " \
      |  ^~~~~~~
RMS/Astrometry/CyFunctions.c: In function ‘__Pyx_InitCachedConstants’:
RMS/Astrometry/CyFunctions.c:8392:290: error: macro "__Pyx_PyCode_New" requires 16 arguments, but only 15 given
 8392 | yx_empty_tuple, __pyx_tuple__8, __pyx_empty_tuple, __pyx_empty_tuple, __pyx_kp_s_RMS_Astrometry_CyFunctions_pyx, __pyx_n_s_subsetCatalog, 70, __pyx_empty_bytes); if (unlikely(!__pyx_codeobj__9)) __PYX_ERR(0, 70, __pyx_L1_error)
      |                                                                                                                                                                ^

RMS/Astrometry/CyFunctions.c:330: note: macro "__Pyx_PyCode_New" defined here
  330 |   #define __Pyx_PyCode_New(a, p, k, l, s, f, code, c, n, v, fv, cell, fn, name, fline, lnos)\
      | 
RMS/Astrometry/CyFunctions.c:8392:33: error: ‘__Pyx_PyCode_New’ undeclared (first use in this function); did you mean ‘PyCode_New’?
 8392 |   __pyx_codeobj__9 = (PyObject*)__Pyx_PyCode_New(5, 0, 14, 0, CO_OPTIMIZED|CO_NEWLOCALS, __pyx_empty_bytes, __pyx_empty_tuple, __pyx_empty_tuple, __pyx_tuple__8, __pyx_empty_tuple, __pyx_empty_tuple, __pyx_kp_s_RMS_Astrometry_CyFunctions_pyx, __pyx_n_s_subsetCatalog, 70, __pyx_empty_bytes); if (unlikely(!__pyx_codeobj__9)) __PYX_ERR(0, 70, __pyx_L1_error)
      |                                 ^~~~~~~~~~~~~~~~
      |                                 PyCode_New
RMS/Astrometry/CyFunctions.c:8392:33: note: each undeclared identifier is reported only once for each function it appears in
RMS/Astrometry/CyFunctions.c:8404:290: error: macro "__Pyx_PyCode_New" requires 16 arguments, but only 15 given
 8404 | pyx_empty_tuple, __pyx_tuple__10, __pyx_empty_tuple, __pyx_empty_tuple, __pyx_kp_s_RMS_Astrometry_CyFunctions_pyx, __pyx_n_s_matchStars, 138, __pyx_empty_bytes); if (unlikely(!__pyx_codeobj__11)) __PYX_ERR(0, 138, __pyx_L1_error)
      |                                                                                                                                                                ^

RMS/Astrometry/CyFunctions.c:330: note: macro "__Pyx_PyCode_New" defined here
  330 |   #define __Pyx_PyCode_New(a, p, k, l, s, f, code, c, n, v, fv, cell, fn, name, fline, lnos)\
      | 
RMS/Astrometry/CyFunctions.c:8416:292: error: macro "__Pyx_PyCode_New" requires 16 arguments, but only 15 given
 8416 | yx_empty_tuple, __pyx_tuple__12, __pyx_empty_tuple, __pyx_empty_tuple, __pyx_kp_s_RMS_Astrometry_CyFunctions_pyx, __pyx_n_s_cyraDecToXY, 211, __pyx_empty_bytes); if (unlikely(!__pyx_codeobj__13)) __PYX_ERR(0, 211, __pyx_L1_error)
      |                                                                                                                                                                ^

RMS/Astrometry/CyFunctions.c:330: note: macro "__Pyx_PyCode_New" defined here
  330 |   #define __Pyx_PyCode_New(a, p, k, l, s, f, code, c, n, v, fv, cell, fn, name, fline, lnos)\
      | 
RMS/Astrometry/CyFunctions.c:8428:292: error: macro "__Pyx_PyCode_New" requires 16 arguments, but only 15 given
 8428 | yx_empty_tuple, __pyx_tuple__14, __pyx_empty_tuple, __pyx_empty_tuple, __pyx_kp_s_RMS_Astrometry_CyFunctions_pyx, __pyx_n_s_cyXYToRADec, 334, __pyx_empty_bytes); if (unlikely(!__pyx_codeobj__15)) __PYX_ERR(0, 334, __pyx_L1_error)
      |                                                                                                                                                                ^

RMS/Astrometry/CyFunctions.c:330: note: macro "__Pyx_PyCode_New" defined here
  330 |   #define __Pyx_PyCode_New(a, p, k, l, s, f, code, c, n, v, fv, cell, fn, name, fline, lnos)\
      | 
RMS/Astrometry/CyFunctions.c: In function ‘__Pyx_CreateCodeObjectForTraceback’:
RMS/Astrometry/CyFunctions.c:10774:5: error: macro "__Pyx_PyCode_New" requires 16 arguments, but only 15 given
10774 |     );
      |     ^
RMS/Astrometry/CyFunctions.c:330: note: macro "__Pyx_PyCode_New" defined here
  330 |   #define __Pyx_PyCode_New(a, p, k, l, s, f, code, c, n, v, fv, cell, fn, name, fline, lnos)\
      | 
RMS/Astrometry/CyFunctions.c:10758:15: error: ‘__Pyx_PyCode_New’ undeclared (first use in this function); did you mean ‘PyCode_New’?
10758 |     py_code = __Pyx_PyCode_New(
      |               ^~~~~~~~~~~~~~~~
      |               PyCode_New
error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
```

dvida, 2020-09-28:
Interesting - could you try it with Python 3.7? What operating systems are you using?

jenyak, 2020-09-28:
I'm using Debian Bullseye. I have already fixed error by replacing [__Pyx_PyCode_New function](https://github.com/CroatianMeteorNetwork/RMS/blob/edacfc725c80fbc78e5585624fed424f00323d47/RMS/Astrometry/CyFunctions.c#L330)

```
  #define __Pyx_PyCode_New(a, p, k, l, s, f, code, c, n, v, fv, cell, fn, name, fline, lnos)\
          PyCode_NewWithPosOnlyArgs(a, p, k, l, s, f, code, c, n, v, fv, cell, fn, name, fline, lnos)
```

with 
```
  #define __Pyx_PyCode_New(a, k, l, s, f, code, c, n, v, fv, cell, fn, name, fline, lnos)\
          PyCode_NewWithPosOnlyArgs(a, 0, k, l, s, f, code, c, n, v, fv, cell, fn, name, fline, lnos)
```
 in every `*.c` file in `RMS` directory. I dont now if `0` is correct value but I was able to install after that.

dvida, 2020-09-28:
Note that the C files are autogenerated by Cython. I guess what happened was that there were pre-compiled C files in the repository that didn't work on your systems, and RMS didn't re-generate them. Once you changed them, it decided to make them again from pyx files. Does it also work when you delete the C files?

jenyak, 2020-09-28:
> Does it also work when you delete the C files?
Yes

jenyak, 2020-09-28:
It didn't re-generate C files when I change them, only when delete them.

dvida, 2020-09-28:
Great - I removed them from the repository now, so we can avoid this problem in the future.

closed 2020-09-28 by dvida