CroatianMeteorNetwork/RMS #54: In SkyFit2, read from Astrometry.net gives 90 degree rotation

tammojan, 2021-02-28:
I think this may be a bug. I used submit to astrometry.net in skyfit2, and it found a solution. However it was 90 degrees rotated.

The FF used is attached, the coordinates used are those of station NL000D.
[FF_NL000F_20210227_201555_422_0228096.fits.gz](https://github.com/CroatianMeteorNetwork/RMS/files/6057325/FF_NL000F_20210227_201555_422_0228096.fits.gz)

dvida, 2021-02-28:
Thanks! Could you try adding 90 deg by yourself in this line: https://github.com/CroatianMeteorNetwork/RMS/blob/7b81456603a2921567d04911993c246301489c85/Utils/SkyFit2.py#L3304
And send a pull request if that fixes it!

tammojan, 2021-02-28:
Sorry, I couldn't reproduce any successful run of an Astrometry solve for skyfit2, it keeps saying 'Waiting too long for the solution!'. So I couldn't test the fix.

dvida, 2021-04-25:
I fixed this in the skyfit2fix branch - I realized that the image uploaded to astrometry.net was transposed.

closed 2021-04-25 by dvida