CroatianMeteorNetwork/RMS #82: Move official parts list from google docs into wiki

jeremy-taylor, 2021-07-30:
The current official parts list is in a google doc https://docs.google.com/document/d/1XBSdrkwrOGPONIn5PBJ7YzH2vr7pUIxW3l8S62BQXEQ/edit - make this part of the wiki and retire the separate list

jeremy-taylor, 2021-08-05:
Purchasing Parts section updated 202010805 ~1700NZST
See notification post : https://groups.io/g/globalmeteornetwork/message/3809

edharman, 2021-08-05:
Hi, great initiative BTW!
wrt your 4th point -
VLC - VideoLan Client is a very capable video media player useful for viewing video clips either locally or remotely
VNC - Virtual Network Computing is a client/server package that allows rendering of a desktop on a remote host onto a local machine - so for example with a Pi with no local display physically connected VNC allows you to say on your network connected laptop mirror the Pi's desktop remotely and launch applications as if you were accessing the 'real' local display.

jeremy-taylor, 2021-08-05:
Thanks Ed!
Yes, I use VLC a fair bit, but or focusing etc you want a 'live' image, not a recorded one. I imagine VNC is more useful, in fact I used VNC on my tablet up while up the ladder to adjust the focus while having the live monitoring script running on the RPi. - like you mentioned in your second description. 

I can't help but thinking that wires got crossed somewhere - but anyway, I don't think this comment about VLC is necessary or fits with the 'tools and parts' bit of the 'building a camera' topic, so I am inclined to remove it - or at least move it somewhere else

jeremy-taylor, 2021-08-06:
When Plate and Bracket option for housing is unavailable, note that the 38x38mm plate is sold separately https://www.aliexpress.com/item/32957085702.html so the "Bracket" option can be purchased along with this separate plate

g7gpr, 2023-01-16:
I'm not sure if this is the correct place but on 

https://globalmeteornetwork.org/wiki/index.php?title=Lens_Options

Seller 1 - https://www.aliexpress.com/item/32876034491.html

is no longer a link to the 6mm lenses.

dvida, 2023-01-16:
Thanks, fixed.

markmac99, 2023-07-10:
all now moved to the wiki as per Radim's work

closed 2023-07-10 by markmac99