CroatianMeteorNetwork/RMS #89: DOC: clarify the Anaconda build item

jeremy-taylor, 2021-08-05:
Ed Harmon volunteered:
https://groups.io/g/globalmeteornetwork/message/3768
see also
https://groups.io/g/globalmeteornetwork/message/3769

apevec, 2024-03-27:
I guess as a first step we would need to reserve the module namespace, unfortunately  as commented in
https://github.com/CroatianMeteorNetwork/RMS/commit/71987b852e720f70f97c167175da78b0c51243ef#r34116469
"RMS" has been taken in pypi by a joke :(

Needs to be tested what does module name change means for the live upgrades in the field.

apevec, 2024-03-27:
To answer the original title "clarify the Anaconda build item". from the  @dvida 's TODO list [1]
* Anaconda build of RMS for easy deployment
technically means to contribute recipe [2] to the conda-forge ?

[1] https://globalmeteornetwork.groups.io/g/main/message/3688
[2] https://conda-forge.org/docs/maintainer/adding_pkgs/