CroatianMeteorNetwork/RMS #93: DOC: focusing strategy

jeremy-taylor, 2021-08-05:
Using Camera params commands to change gain to assist in focus and pointing setup

jeremy-taylor, 2021-08-05:
If you're on a Pi4 you can do this using the CameraControl module.

python -m Utils.CameraControl SetParam Camera Style 2
python -m Utils.CameraControl SetParam Camera DayNightColor 1
python -m Utils.CameraControl SetParam Camera ElecLevel 30
python -m Utils.CameraControl SetParam Camera ExposureParam LeastTime 100
python -m Utils.CameraControl SetParam Camera GainParam Gain 30

And to set it back again, use 1, 2, 100, 40000 and 60 respectively.

Take a look at Scripts/RMS_SetCameraParams.sh to see the other things you can alter.

jeremy-taylor, 2021-08-05:
first command turns out actually to be python -m Utils.CameraControl SetParam Camera Style type2 and to undo it python -m Utils.CameraControl SetParam Camera Style type1

markmac99, 2023-07-10:
I've added this info to the wiki
https://globalmeteornetwork.org//wiki/index.php?title=Focusing_your_camera#RE-Focusing_Later

closed 2023-07-10 by markmac99