CroatianMeteorNetwork/RMS #104: Request - save .config in the CapturedFiles observation directory.

g7gpr, 2021-09-27:
It would be useful to have the .config file used for the observing session saved in the CapturedFiles observation directory.

dvida, 2021-09-27:
It is already saved to ArchivedFiles - does that directory have everything you need, or do you have a specific use case where you need to use CapturedFiles?

g7gpr, 2021-09-27:
By saving it into CapturedFiles I can have a single archive which contains all the raw data and the associated .config from an observation session. The use case for this to be able to set up a camera in one location, run a test observation, keep the reference CapturedFiles and then deploy the camera to another location.  Any quirks in the new location can then be checked against the test observation. Manually copying the .config across from ArchivedFiles is my workaround, but there is always the possibility of making an error.

g7gpr, 2021-10-08:
Thank you. This does exactly what I need.

closed 2021-10-08 by g7gpr