CroatianMeteorNetwork/RMS #159: SkyFit2 error message: ModuleNotFoundError: No module named 'pyqtgraph'

TobyOnGithub, 2023-03-05:
I installed RMS from scratch on my Ubuntu Mate (Release 22.04.2 LTS (Jammy Jellyfish) 64-bit). Installation took around 70 - 80 minutes. The I tried to run skyfit2 using command "python -m Utils.SkyFit2 &". Then I got the error message "ModuleNotFoundError: No module named 'pyqtgraph'". Then I installed the module using "pip install pyqtgraph".

markmac99, 2023-03-16:
RMS isn't really supported on Ubuntu 22.04 as there are a number of issues with OpenCV and other components. Its on a list of issues to investigate but there's no ETA. If you have any python skills then your assistance in debugging would be welcome

dvida, 2023-03-16:
I think the issue here was with the instructions., these here: https://globalmeteornetwork.org/wiki/index.php?title=Installation_for_Linux

I'll fix them.

closed 2023-03-16 by dvida