CroatianMeteorNetwork/RMS #179: Archive detections fails if running Reprocess on a directory with no FITS files

g7gpr, 2023-06-24:
2023/06/23 22:51:27-INFO-Reprocess-line:511 - Archiving detections to /home/au001a/RMS_data/ArchivedFiles/AU001A_20230623_094050_841901
2023/06/23 22:51:27-INFO-ArchiveDetections-line:171 - Generating thumbnails...
2023/06/23 22:51:28-INFO-ArchiveDetections-line:192 - Generating a stack of detections...
2023/06/23 22:51:28-INFO-ArchiveDetections-line:219 - Captured stack could not be saved!
2023/06/23 22:51:28-INFO-ArchiveDetections-line:237 - Detected stack could not be saved!
Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/au001a/source/RMS/RMS/StartCapture.py", line 1068, in <module>
    night_archive_dir = runCapture(config, duration=duration, nodetect=cml_args.nodetect, \
  File "/home/au001a/source/RMS/RMS/StartCapture.py", line 512, in runCapture
    night_archive_dir, archive_name, _ = processNight(night_data_dir, config, \
  File "/home/au001a/source/RMS/RMS/Reprocess.py", line 514, in processNight
    archive_name = archiveDetections(night_data_dir, night_archive_dir, ff_detected, config, \
  File "/home/au001a/source/RMS/RMS/ArchiveDetections.py", line 253, in archiveDetections
    archive_name = archiveDir(captured_path, file_list, archived_path, archive_name, \
  File "/home/au001a/source/RMS/RMS/Misc.py", line 151, in archiveDir
    shutil.copy2(file_name, os.path.join(dest_dir, os.path.basename(file_name)))
  File "/usr/lib/python3.9/shutil.py", line 435, in copy2
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "/usr/lib/python3.9/shutil.py", line 264, in copyfile
    with open(src, 'rb') as fsrc, open(dst, 'wb') as fdst:
FileNotFoundError: [Errno 2] No such file or directory: '/home/au001a/RMS_data/CapturedFiles/AU001A_20230623_094050_841901/AU001A_20230623_094050_841901_timelapse.mp4'

markmac99, 2023-07-10:
Not sure this is a bug since no detection can be done on a folder with no FITS files ! 
However, it should certainly fail more gracefully.  The error is arising because there's no timelapse, so i'll examine the code and put in a fix.

markmac99, 2023-07-30:
I've put in try/except around the lines that copy files to ArchiveDir, and added logging to the normal RMS logfile. Tested as best i can as i have not had an occasion where no files were created, but on the other hand it has not broken normal operation.

g7gpr, 2023-07-30:
I'll be able to test it during the next 7 days. Thanks for investigating.

PR #185 (2023-07-31): Fix for issues 179 and 181
NOW TESTED WITHOUT ISSUES

1) ExtractStars doesn't log any info, so there is no warning if TOO MANY stars were detected. This update sends logging to the normal RMS logfile so that at least we can see when the problem arose. Closes #181 

2) archiveDir crashes if the timelapse or any other extra_files entry is missing. This can happen if zero FITS files were captured, a rare state but apparently one that end-users have encountered. I've put in try/except around the lines that copy files to ArchiveDir, and added logging to the normal RMS logfile. Fixes #179