CroatianMeteorNetwork/RMS #210: Updates to CameraControl

Vasiliy1992, 2023-09-20:
It was noticed that the command does not work on Raspberry Pi 3B+:
 
         python -m Utils.CameraControl reboot

Can you fix it? This feature is very necessary.

opp4ce, 2023-09-21:
I would like to clarify, this command does not work at Rpi 3B+

`python -m Utils.CameraControl27 reboot`

There is no feedback in the console. If ping is started in another terminal window, then there are no interruptions in the execution of ping command.

Vasiliy1992, 2023-09-21:
```
(vRMS) pi@RMS-RU0001:~/source/RMS $ python -m Utils.CameraControl reboot
Loading the default config!
rebooting, please wait....
error executing command - probably not supported
(vRMS) pi@RMS-RU0001:~/source/RMS $ python -m Utils.CameraControl27 reboot
(vRMS) pi@RMS-RU0001:~/source/RMS $ 

```