CroatianMeteorNetwork/RMS #231: Runtime error Debian 12 NUMPY

oldjiberjaber, 2023-11-12:
I'm in the process of moving to a faster PC for my RMS installation, RMS is presently working on Deb11 but I've built the new system with Deb12 and I am running in to the following issue which looks like it needs a directive adding to manage which API is in use - but I have no idea where to put it and it would probably be overwritten with teh next update from GIT :) 

I'm assuming there isn't a library missing, any ideas where and what in order to resolve please? 

```
(vRMS) jason@shed1:~/source/RMS$ ./Scripts/RMS_StartCapture.sh
Starting RMS...

If you need to update the RMS config file, you can do it now.
Any changes to the config file will be read only after this script is started again or the Pi is rebooted.

Compiling /home/jason/source/RMS/RMS/Routines/MorphCy.pyx because it changed.
[1/1] Cythonizing /home/jason/source/RMS/RMS/Routines/MorphCy.pyx
/home/jason/vRMS/lib/python3.11/site-packages/Cython/Compiler/Main.py:381: FutureWarning: Cython directive 'language_level' not set, using '3str' for now (Py3). This has changed from earlier releases! File: /home/jason/source/RMS/RMS/Routines/MorphCy.pyx
  tree = Parsing.p_module(s, pxd, full_module_name)
In file included from /home/jason/vRMS/lib/python3.11/site-packages/numpy/core/include/numpy/ndarraytypes.h:1929,
                 from /home/jason/vRMS/lib/python3.11/site-packages/numpy/core/include/numpy/ndarrayobject.h:12,
                 from /home/jason/vRMS/lib/python3.11/site-packages/numpy/core/include/numpy/arrayobject.h:5,
                 from /home/jason/.pyxbld/temp.linux-x86_64-cpython-311/home/jason/source/RMS/RMS/Routines/MorphCy.c:1211:
/home/jason/vRMS/lib/python3.11/site-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:17:2: warning: #warning "Using deprecated NumPy API, disable it with " "#define NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
   17 | #warning "Using deprecated NumPy API, disable it with " \
      |  ^~~~~~~
/home/jason/vRMS/lib/python3.11/site-packages/Cython/Compiler/Main.py:381: FutureWarning: Cython directive 'language_level' not set, using '3str' for now (Py3). This has changed from earlier releases! File: /home/jason/source/RMS/RMS/Routines/BinImageCy.pyx
  tree = Parsing.p_module(s, pxd, full_module_name)
In file included from /home/jason/vRMS/lib/python3.11/site-packages/numpy/core/include/numpy/ndarraytypes.h:1929,
                 from /home/jason/vRMS/lib/python3.11/site-packages/numpy/core/include/numpy/ndarrayobject.h:12,
                 from /home/jason/vRMS/lib/python3.11/site-packages/numpy/core/include/numpy/arrayobject.h:5,
                 from /home/jason/.pyxbld/temp.linux-x86_64-cpython-311/home/jason/source/RMS/RMS/Routines/BinImageCy.c:1205:
/home/jason/vRMS/lib/python3.11/site-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:17:2: warning: #warning "Using deprecated NumPy API, disable it with " "#define NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
   17 | #warning "Using deprecated NumPy API, disable it with " \
      |  ^~~~~~~
RuntimeError: The new DType API is currently in an exploratory phase and should NOT be used for production code.  Expect modifications and crashes!  To experiment with the new API you must set `NUMPY_EXPERIMENTAL_DTYPE_API=1` as an environment variable.
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/jason/source/RMS/RMS/StartCapture.py", line 43, in <module>
    from RMS.Compression import Compressor
  File "/home/jason/source/RMS/RMS/Compression.py", line 28, in <module>
    from RMS.VideoExtraction import Extractor
  File "/home/jason/source/RMS/RMS/VideoExtraction.py", line 26, in <module>
    from RMS.DetectionTools import loadImageCalibration, binImageCalibration
  File "/home/jason/source/RMS/RMS/DetectionTools.py", line 15, in <module>
    from RMS.Math import vectNorm
  File "/home/jason/source/RMS/RMS/Math.py", line 4, in <module>
    from numpy.core.umath_tests import inner1d
  File "/home/jason/vRMS/lib/python3.11/site-packages/numpy/core/umath_tests.py", line 13, in <module>
    from ._umath_tests import *
RuntimeError: cannot load _umath_tests module.

```

g7gpr, 2024-06-18:
Did you overcome this?

oldjiberjaber, 2025-02-13:
Nope - I just gave up and I've not touched it since, I've a working version on an older distro for now, but I expect the latest RMS might have caught up and resolved itself?