CroatianMeteorNetwork/RMS #236: No automatically detected stars

LeoBosse, 2023-12-05:
Hello,
I just installed skyfit2 on my ubuntu 22.04 laptop following [the wiki](https://globalmeteornetwork.org/wiki/index.php?title=Installation_for_Linux), and started following [this youtube tutorial.](https://www.youtube.com/watch?v=ao3J9Jf0iLQ) 
Once opened, I can play with the astrometry parameters to align approximately the catalog stars to my image. But there are no green circles appearing automatically, and pressing C does nothing. The "fit" button is greyed out and cannot be clicked.
I tried to play a little with the config file parameters under [StarExtraction] without success. I have 9 .fits images with background around 500 counts, while the stars have a few 1000 counts. The limiting mag is a least 7, and image dim are ~30x40degrees for 4144x2800 pixels. The FPS is 0.2 (5 sec exposures). 

Why is that? Did I miss something during install? Is it because of a low SNR? Is there a way to force the detection?

Thanks a lot in advance for your help!

dvida, 2023-12-05:
The star detector requires fine-tuning and it only works on RMS-type FF files. But you don't need any detected stars to make a good plate. Just centroid the stars yourself and you'll be good to go.
Also, because the exposure is so long, I recommend you analyze images individually (one per folder), otherwise you might get long star trails in the combined image.

LeoBosse, 2023-12-06:
Wow, thank you so much for such a quick answer! Ok, I'll see how to do by hand. Do you know if there is a way to convert fits to RMS-type FF files?

LeoBosse, 2023-12-07:
Thanks for your help, I managed to load our fits images with detected stars! Playing with the FFfits.py read function, I had to adapt the header parameters to ours as well as the HDUList, set full_filename=True by default and it works like a charm!