CroatianMeteorNetwork/RMS #252: RMS stops image capturing during strong wind and camera vibration

aitov, 2024-01-12:
Hi, I have camera inside large box and during strong wind some vibration is happening (as result blurry fits images ). 
But noticed if wind is very strong: RMS just stops image capturing without any error in log and you just missed the most of nigh.

It was reproduced 3rd time over 5 month.
Example of log below:

```
2024/01/11 22:36:55-INFO-BufferedCapture-line:474 - New block of raw frames available for compression with starting time: 1705012595.1614099
2024/01/11 22:36:55-INFO-BufferedCapture-line:331 - Grabbing a new block of 256 frames...
2024/01/11 22:36:55-DEBUG-Compression-line:274 - Compressing frame block with start time at: 1705012595.1614099
2024/01/11 22:37:02-DEBUG-Compression-line:295 - Compression time: 7.219 s
2024/01/11 22:37:02-DEBUG-Compression-line:302 - Saving time: 0.068 s
2024/01/11 22:37:03-INFO-DetectionTools-line:61 - Loaded mask: mask.bmp
2024/01/11 22:37:03-DEBUG-Compression-line:319 - Extractor started for: UA0004_20240111_223635_161_0110336
2024/01/11 22:37:03-INFO-DetectStarsAndMeteors-line:63 - Running detection on file: FF_UA0004_20240111_223635_161_0110336.fits
2024/01/11 22:37:03-INFO-DetectionTools-line:61 - Loaded mask: mask.bmp
2024/01/11 22:37:03-INFO-Compression-line:332 - Added file for detection: FF_UA0004_20240111_223635_161_0110336.fits
2024/01/11 22:37:03-DEBUG-VideoExtraction-line:285 - [UA0004_20240111_223635_161_0110336] time for thresholding and subsampling: 0.4954211711883545s
2024/01/11 22:37:03-DEBUG-VideoExtraction-line:289 - [UA0004_20240111_223635_161_0110336] nothing found, not extracting anything 1
2024/01/11 22:37:05-INFO-ExtractStars-line:186 - extracted 39 stars from FF_UA0004_20240111_223635_161_0110336.fits
2024/01/11 22:37:05-INFO-DetectStarsAndMeteors-line:89 - Detected stars: 1
2024/01/11 22:49:24-INFO-EventMonitor-line:2007 - Event monitor check completed
2024/01/11 23:19:25-INFO-EventMonitor-line:2007 - Event monitor check completed
2024/01/11 23:49:25-INFO-EventMonitor-line:2007 - Event monitor check completed
2024/01/12 00:19:26-INFO-EventMonitor-line:1067 - Added event at 20231228_054329 to the database
2024/01/12 00:19:26-INFO-EventMonitor-line:1775 - Checks on trajectories for event at 20231228_054329
2024/01/12 00:19:26-INFO-EventMonitor-line:1483 - Using .config as .config file name
2024/01/12 00:19:26-INFO-EventMonitor-line:1785 - No files for event - marking 20231228_054329 as processed
2024/01/12 00:19:26-INFO-EventMonitor-line:1093 - Event at 20231228_054329 marked as processed
2024/01/12 00:19:26-INFO-EventMonitor-line:1921 - 1 event was processed, EventMonitor work completed
2024/01/12 00:19:26-INFO-EventMonitor-line:2007 - Event monitor check completed
2024/01/12 00:49:27-INFO-EventMonitor-line:2007 - Event monitor check completed
2024/01/12 01:19:27-INFO-EventMonitor-line:2007 - Event monitor check completed
2024/01/12 01:49:28-INFO-EventMonitor-line:2007 - Event monitor check completed
2024/01/12 02:19:28-INFO-EventMonitor-line:2007 - Event monitor check completed
2024/01/12 02:49:29-INFO-EventMonitor-line:2007 - Event monitor check completed
2024/01/12 03:19:30-INFO-EventMonitor-line:2007 - Event monitor check completed
2024/01/12 03:49:31-INFO-EventMonitor-line:2007 - Event monitor check completed
2024/01/12 04:19:31-INFO-EventMonitor-line:2007 - Event monitor check completed
2024/01/12 04:49:32-INFO-EventMonitor-line:2007 - Event monitor check completed
2024/01/12 05:19:33-INFO-EventMonitor-line:2007 - Event monitor check completed
2024/01/12 05:20:41-DEBUG-StartCapture-line:390 - Stopping capture...
2024/01/12 05:20:42-INFO-BufferedCapture-line:96 - Joining capture...
2024/01/12 05:21:42-INFO-BufferedCapture-line:106 - Terminating capture...
2024/01/12 05:21:42-DEBUG-StartCapture-line:393 - Capture stopped
```

dvida, 2024-01-12:
The issue is probably a dropped connection to the camera or some other video issue. Have you tried connecting to the camera in the morning after failure?

aitov, 2024-01-13:
Hi, camera connected directly to RPi (MIPI camera). 
According to the log record:

`
2024/01/12 05:21:42-INFO-BufferedCapture-line:106 - Terminating capture...
`

It stuck in main loop during grabbing frames and I found only one possible place for deadlock:
`
ret, frame = device.read()
` 

So it could be opencv or hardware issue

In opencv version 4.8 added timeout properties : CAP_PROP_READ_TIMEOUT_MSEC
```
timeout in milliseconds for reading from a video capture (applicable for FFmpeg and GStreamer back-ends only)
```
Probably it might help for reconnection to camera

aitov, 2024-01-13:
Tried to reproduce on new device -  can't reproduce. With vibration, movements and hits still continue processing. Need check on original device, seems like hardware issue.