CroatianMeteorNetwork/RMS #270: Set a maximum time between archived fits files

g7gpr, 2024-02-18:
RMS stations provide good quality images of the sky. A feature could be added to provide a maximum time difference between fits files, so that if no meteor was detected for a greater period than that time, fits files would be brough in from the night_captured_dir. 

In this way, for a small additional number of fits files, a comprehensive record of the night sky from each station could be held at gmn.uwo.ca

markmac99, 2024-07-31:
I'd caution against this. I feel it would be "mission creep": RMS's purpose is to track meteors rather than the night sky, and we already track sky conditions with a combination of the timelapse and the sky brightness data.

dvida, 2024-08-01:
We're planning to institute a few changes in the code which will help track contrails and will produce captures of single frames at regular intervals, which might solve what this proposal aims to do.

closed 2024-08-01 by g7gpr