CroatianMeteorNetwork/RMS #273: Random camera reboots

Cybis320, 2024-03-03:
Regarding the camera auto reboot at 15:00, The cameras don't have batteries. When unpowered, they subsequently resume running the clock from whatever last time was stored. They lose about ~30 sec per day from daily reboots on top of other power events. If one set the clock with CMS on a PC then leave the camera unpowered for some time, it's impossible to predict when the camera will reboot.

They are a couple potential solutions to this. One could enable the camera NTP feature so that its clock gets updated regularly, or the auto-reboot could be disabled and a script containing a reboot command run before or after each run. This would work for any time zone.

Personally, I've disabled auto reboot and I'm running a script with a reboot before each run. It works well for me because I'm running a night script and a day script with different settings.

What are your thoughts on implementing this? I agreed that losing ~50sec of data a night is really not a big deal but it's just cleaner not to have these random reboots.

markmac99, 2024-04-16:
Yes, the cameras do lose time if powered off.  

If you experience frequent power outages then the simplest solution is to run a script that executes ```python -m Utils.CameraControl CameraTime set yyyymmdd_hhmmss``` whenever the pi boots.  The daily reboot setting should then work as configured. This would have the advantage of ensuring the camera's internal clock was correct (though we don't actually use it). 

Configuring NTP is not straightforward, since the camera modules don't usually have internet access (they're on a private subnet connected only to the Pi). So one would need to run an ntp server on the Pi itself which would certainly be possible but is probably outside RMS's scope. In my own case i run an ntp server on my home network and my cameras are connected via a router, but most users don't have this option. 

I'd be happy to add a specific Script to set the camera clock, which users could optionally execute as desired?

Cybis320, 2025-05-31:
Fixed in prerelease by reseting camera time at camera mode switching.

closed 2025-05-31 by Cybis320