CroatianMeteorNetwork/RMS #285: Add setColorParams to cameraControl

markmac99, 2024-04-15:
As noted by Milan, we can read but not set the contrast, brightness, gain, hue saturation and whitebalance. 
Should add SetColorParams to set these.

satmonkey, 2024-04-16:
there is SetColor command, which is used in RMS_SetCameraParams.sh:
python -m Utils.CameraControl SetColor 100,50,50,50,0,0
However, by this, brightness is set to 100%, should be 50% (50,50,50,50,0,0)

Cybis320, 2024-04-16:
The 100 brightness setting pins the camera's gain to the maximum. It's also the reason the image goes to full white at sunset and sunrise. If it's set to 50, there's a risk that the gain will fluctuate to maintain a 50% average value. I set mine to 70, which seems to pin the gain to max during astronomical night but prevents recording just white frames during twilight.

satmonkey, 2024-04-16:
I think brightness has nothing to do with the gain here. It is just post-process setting, which just further down narrows the dynamic range.

Cybis320, 2024-04-16:
Argh, you are absolutely right. ElecLevel is the one I was thinking about. 
I should not have been typing while getting off the plane, sorry.

markmac99, 2024-04-16:
As milan says this is a different set of settings. Its the ones that appear in CMS when you right click and select "Color config" rather than the ones on the Device Config->System->Camera Param. dialog

@satmonkey - you're right, SetColor does set these values but the name is inconsistent with the command to retrieve data, which i think is a source of confusion (even to me, who wrote it...! ). I will create an alias SetColorParams so the Get and Set are consistent.

ps - setting brightness to 100: I experiemented with various levels and found this gave me best results for night-time capture in my location. That said, i don't think it makes a *huge* difference. UK0006 has been set to 30 for a few weeks, whereas its neighbour UK000F has been set to 100, and both have detected roughly the same number of meteors. The main difference seems to be the background greyness of the image - higher brightness seems to lead to blacker background. Compare the top two images here https://markmcintyreastro.co.uk/cameradata/ which are for the two above cameras. UK0006, the one set to 30, is a little greyer than the ones set to 100 (the other three) though the difference is small.

satmonkey, 2024-04-16:
Brightness - yes, depending on the particular sky polution level, it may look more suitable for cosmetic reasons.
But, here, higher brightness yields brighter background, strange... I have just experimented yesterday with:
SetColor 50,50,50,50,0,0
and
SetColor 100,50,50,50,0,0
First, I saw the 100% level (caused by running the script earlier), when I noticed the drop in meteor count at two stations, and found out the sky is unusually bright. I have put that back, and since that, everyting is fine, again.