CroatianMeteorNetwork/RMS #288: crash when processing data: "AttributeError: module 'numpy' has no attribute 'bool'."

pioupus, 2024-05-04:
excerpt from the log:
2024/05/04 16:07:36-ERROR-StartCapture-line:708 - An error occurred when trying to reprocess partially processed data!
2024/05/04 16:07:36-ERROR-StartCapture-line:709 - AttributeError("module 'numpy' has no attribute 'bool'.\n`np.bool` was a deprecated alias for the builtin `bool`. To avoid this error in existing code, use `bool` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.bool_` here.\nThe aliases was originally deprecated in NumPy 1.20; for more details and guidance see the original release note at:\n    https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations")
2024/05/04 16:07:36-ERROR-StartCapture-line:710 - ['Traceback (most recent call last):\n', '  File "/home/rms/source/RMS/RMS/StartCapture.py", line 687, in processIncompleteCaptures\n    night_archive_dir, archive_name, detector = processNight(captured_dir_path, config)\n', '  File "/home/rms/source/RMS/RMS/Reprocess.py", line 225, in processNight\n    recalibrated_platepars = recalibrateIndividualFFsAndApplyAstrometry(night_data_dir, \\\n', '  File "/home/rms/source/RMS/RMS/Astrometry/ApplyRecalibrate.py", line 677, in recalibrateIndividualFFsAndApplyAstrometry\n    recalibrated_platepars = recalibratePlateparsForFF(\n', '  File "/home/rms/source/RMS/RMS/Astrometry/ApplyRecalibrate.py", line 439, in recalibratePlateparsForFF\n    test_platepar = alignPlatepar(\n', '  File "/home/rms/source/RMS/RMS/Astrometry/FFTalign.py", line 284, in alignPlatepar\n    res = findStarsTransform(config, calstars_coords, catalog_xy, show_plot=show_plot)\n', '  File "/home/rms/source/RMS/RMS/Astrometry/FFTalign.py", line 150, in findStarsTransform\n    res = imreg_dft.imreg.similarity(img_ref, img_mov)\n', '  File "/home/rms/vRMS/lib/python3.9/site-packages/imreg_dft/imreg.py", line 376, in similarity\n    bgval = utils.get_borderval(im1, 5)\n', '  File "/home/rms/vRMS/lib/python3.9/site-packages/imreg_dft/utils.py", line 705, in get_borderval\n    mask = np.zeros_like(img, dtype=np.bool)\n', '  File "/home/rms/vRMS/lib/python3.9/site-packages/numpy/__init__.py", line 324, in __getattr__\n    raise AttributeError(__former_attrs__[attr])\n', "AttributeError: module 'numpy' has no attribute 'bool'.\n`np.bool` was a deprecated alias for the builtin `bool`. To avoid this error in existing code, use `bool` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.bool_` here.\nThe aliases was originally deprecated in NumPy 1.20; for more details and guidance see the original release note at:\n    https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations\n"]
2024/05/04 16:07:36-INFO-StartCapture-line:1053 - Waiting 5:14:51.684344 to start recording for 7.361 hrs

dvida, 2024-05-04:
We have a fix ready, we're just waiting for the eta-Aquariids to pass to deploy it.
Do fix it manually, run:
pip uninstall imreg_dft
pip install --force-reinstall git+https://github.com/matejak/imreg_dft@master#egg=imreg_dft

pioupus, 2024-05-04:
Thank you for your help.
the proposed command for installing the package didnt work. For some reason pip could not access github. But the following line worked for me. Hopefully it did the same thing:

pip install --force-reinstall git+https://github.com/matejak/imreg_dft.git

dvida, 2024-05-05:
I just pushed a fix into master.

closed 2024-05-05 by dvida

dvida, 2024-05-09:
```fatal: unable to access 'https://github.com/matejak/imreg_dft/': Couldn't connect to server```
It looks like an internet connection issue, can you try again?