CroatianMeteorNetwork/RMS #291: np.bool deprecated

g7gpr, 2024-06-03:
Seen since updating a non-gui, one camera per username system to Debian 12

```
2024/06/03 13:47:28-ERROR-StartCapture-line:708 - An error occurred when trying to reprocess partially processed data!
2024/06/03 13:47:28-ERROR-StartCapture-line:709 - AttributeError("module 'numpy' has no attribute 'bool'.\n`np.bool` was a deprecated alias for the builtin `bool`. To avoid this error in existing code, use `bool` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.bool_` here.\nThe aliases was originally deprecated in NumPy 1.20; for more details and guidance see the original release note at:\n    https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations")
```
and also

```
2024/06/03 13:47:28-ERROR-StartCapture-line:710 - ['Traceback (most recent call last):\n', '  File "/home/au0028/source/RMS/RMS/StartCapture.py", line 687, in processIncompleteCaptures\n    night_archive_dir, archive_name, detector = processNight(captured_dir_path, config)\n                                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^\n', '  File "/home/au0028/source/RMS/RMS/Reprocess.py", line 225, in processNight\n    recalibrated_platepars = recalibrateIndividualFFsAndApplyAstrometry(night_data_dir, \\\n                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^\n', '  File "/home/au0028/source/RMS/RMS/Astrometry/ApplyRecalibrate.py", line 677, in recalibrateIndividualFFsAndApplyAstrometry\n    recalibrated_platepars = recalibratePlateparsForFF(\n                             ^^^^^^^^^^^^^^^^^^^^^^^^^^\n', '  File "/home/au0028/source/RMS/RMS/Astrometry/ApplyRecalibrate.py", line 439, in recalibratePlateparsForFF\n    test_platepar = alignPlatepar(\n                    ^^^^^^^^^^^^^^\n', '  File "/home/au0028/source/RMS/RMS/Astrometry/FFTalign.py", line 284, in alignPlatepar\n    res = findStarsTransform(config, calstars_coords, catalog_xy, show_plot=show_plot)\n          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^\n', '  File "/home/au0028/source/RMS/RMS/Astrometry/FFTalign.py", line 150, in findStarsTransform\n    res = imreg_dft.imreg.similarity(img_ref, img_mov)\n          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^\n', '  File "/home/au0028/vRMS/lib/python3.11/site-packages/imreg_dft/imreg.py", line 376, in similarity\n    bgval = utils.get_borderval(im1, 5)\n            ^^^^^^^^^^^^^^^^^^^^^^^^^^^\n', '  File "/home/au0028/vRMS/lib/python3.11/site-packages/imreg_dft/utils.py", line 705, in get_borderval\n    mask = np.zeros_like(img, dtype=np.bool)\n                                    ^^^^^^^\n', '  File "/home/au0028/vRMS/lib/python3.11/site-packages/numpy/__init__.py", line 324, in __getattr__\n    raise AttributeError(__former_attrs__[attr])\n', "AttributeError: module 'numpy' has no attribute 'bool'.\n`np.bool` was a deprecated alias for the builtin `bool`. To avoid this error in existing code, use `bool` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.bool_` here.\nThe aliases was originally deprecated in NumPy 1.20; for more details and guidance see the original release note at:\n    https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations\n"]
```

dvida, 2024-06-03:
The issue is with the imreg_dft library, run these commands and issues should go away:
pip uninstall imreg_dft
pip install --force-reinstall git+https://github.com/matejak/imreg_dft@master#egg=imreg_dft

How did you install the system in the first place?

g7gpr, 2024-06-03:
Using the instruction at 

https://globalmeteornetwork.org/wiki/index.php?title=Installation_for_Linux

However, I believe this issue is not related to this repo, so I'll close with apologies.

closed 2024-06-03 by g7gpr