CroatianMeteorNetwork/RMS #292: Plotting time intervals fails

dvida, 2024-06-03:
`2024/06/03 22:11:11-DEBUG-Reprocess-line:408 - Plotting timestamp interval failed with message:
TypeError('unsupported format string passed to NoneType.__format__')
2024/06/03 22:11:11-DEBUG-Reprocess-line:409 - ['Traceback (most recent call last):\n', '  File "/home/lcam/source/RMS/RMS/Reprocess.py", line 401, in processNight\n    log.info(\'Timestamp Intervals Analysis: Jitter Quality: {:.1f}%, Dropped Frame Rate: {:.1f}%\'.format(\n             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^\n', 'TypeError: unsupported format string passed to NoneType.__format__\n']`

Cybis320, 2024-06-04:
This would typically happen when there aren't enough data points to compute jitter and dropped frames. Is it happening outside of that scenario?

dvida, 2024-06-04:
It is not, this was a very short capture run. I caught it only by chance by looking at the scrolling log. Would it be possible to handle this gracefully so it doesn't error out?

Thanks!

closed 2024-06-07 by dvida