CroatianMeteorNetwork/RMS #306: Unhandled exception in CMN_binviewer when doing manual confirmations

g7gpr, 2024-07-09:
After completing manual confirmations the following unhandled exception was raised.

```
2024/07/10 05:27:24-CRITICAL: Traceback (most recent call last):
  File "/home/david/source/cmn_binviewer/CMN_binViewer.py", line 3877, in __call__
    return self.func(*args)
  File "/home/david/source/cmn_binviewer/CMN_binViewer.py", line 3317, in confirmationEnd
    newufoData = self.updateUFOData(FTPdetectinfoExport, ufoData)
  File "/home/david/source/cmn_binviewer/CMN_binViewer.py", line 917, in updateUFOData
    newufoData = np.insert(tmparr, 0, ufoData[0])
  File "/home/david/vBinviewer/lib/python3.9/site-packages/numpy/lib/function_base.py", line 5513, in insert
    values = array(values, copy=False, ndmin=arr.ndim, dtype=arr.dtype)
ValueError: could not convert string to float: 'Ver,Y,M,D,h,m,s,Mag,Dur,Az1,Alt1,Az2,Alt2, Ra1, Dec1, Ra2, Dec2,ID,Long,Lat,Alt,Tz\n'
```

The dataset is saved at 

```
https://1drv.ms/u/c/6165251cc10ab7b0/ET0im2qgLPFIpNftLKlf_Z0BGLhQTR8eyUyTV-bJouSNyA?e=T2xVMG
```

markmac99, 2024-07-31:
I can't reproduce this issue. I downloaded the dataset and successfully ran the confirmation process with BinViewer 3.37 (the latest release). 
Please check you have the latest version installed, and then please explain the exact steps required to recreate the problem.

markmac99, 2024-07-31:
Also please note that binviewer has its own repository and codebase, so issues should be raised there.

g7gpr, 2024-08-01:
Thanks, I though the repos had been combined. I'll update to the latest version and report back.

closed 2024-08-01 by g7gpr