CroatianMeteorNetwork/RMS #319: kht module detection is failing when python minor version > 9

g7gpr, 2024-07-31:
In ConfigReader.py find BinaryPath generates a python version such as '3.11' using the code at 

https://github.com/CroatianMeteorNetwork/RMS/blob/3a20e04d8fe8104b02af82b3fc34d3e00ace2bd4/RMS/ConfigReader.py#L117

however the kht module is found in a directory named

```
lib.linux-x86_64-cpython-311
```

which does not include a dot. This causes the kht module not to be found, leading to the kht_wrapper bug. 

Earlier version of python/cython placed the binary files in a directory with a dot separator

```
lib.linux-x86_64-3.7
```

I suspect that this could be quickly fixed by giving an explicit path in the .config file, working on a patch on 

https://github.com/CroatianMeteorNetwork/RMS/blob/3a20e04d8fe8104b02af82b3fc34d3e00ace2bd4/RMS/ConfigReader.py#L117

dvida, 2024-07-31:
I believe I encountered the same issue and the solution was to purge all previous builds, but I might be wrong.

dvida, 2024-07-31:
These are the names on build dirs on a Pi3 running Python 2:
[image]

I belive the build script already tries to delete a build directory, but that may fail if e.g. the previous build was done with sudo. But Dave's approach is the true way forward, one a check for Py2 is added.

g7gpr, 2024-08-07:
Merged into prerelease, so closing now.

closed 2024-08-07 by g7gpr