CroatianMeteorNetwork/RMS #322: 'bad marshal data' Python error

sammmelg, 2024-07-31:
Hello,
An error cropped up on one of my stations (USL017) and Denis advised me to open an issue describing the error and how I solved it.

When RMS_StartCapture initiated, several lines would run before the Python error 'ValueError: bad marshal data (unknown type code' would pop up and cause the script to crash. I found out through a google search that this error can be cause by corrupted .pyc files. To solve the error, I went thru the traceback and deleted .pyc files for the packages that showed up on the traceback. I also restarted RMS_StartCapture.sh after each one, to see if the correct .pyc file had been deleted. Finally, once all of the corrupted files had been deleted, StartCapture came up with no issues and has been running successfully for about a week now.

Attached is a screenshot of the error that was showing up.

Sam Hemmelgarn

[image]

dvida, 2024-07-31:
Agreed, would you mind addressing this? Perhaps there already is a mechanism in setup.py that we haven't implemented, it would only make sense.

markmac99, 2024-07-31:
You could add pyclean to the requirements, then run 
```
pyclean -v --debris all 
```
This would be a pretty safe way to clear out any cached and other temporary files.

closed 2024-08-08 by Cybis320