CroatianMeteorNetwork/RMS #323: Better pruning of CapturedFiles folders for multicam setups

markmac99, 2024-07-31:
Currently, RMS prunes CapturedFiles folders in order to free up space for the next night's run. This works fine on a single-camera station. However, on multi-cam stations there are multiple RMS instances running and they compete to clear space. This can lead to one camera being more aggressively pruned than the others. It'd be better if the capdirs were more evenly pruned.

The request is to add a parameter "cap_dirs_to_keep" similar to the parameter arch_dirs_to_keep. so specify how many to keep.

g7gpr, 2024-07-31:
I can take this.

g7gpr, 2024-08-02:
Tested, apparently working.

g7gpr, 2024-08-12:
```
2024/08/12 13:16:15-INFO-DeleteOldObservations-line:584 -

--------------------------------------------
Directory quotas before management
--------------------------------------------
Space used
   Total space used for RMS_data : 0099.41GB
            bz2 files space used : 0001.97GB
 archived directories space used : 0004.52GB
              total for archives : 0006.48GB
   Captured directory space used : 0092.63GB
Quotas allowed
        Total quota for RMS_data : 0100.00GB
                  bz2 file quota : 0003.00GB
      Archived directories quota : 0005.00GB
          Remaining for captured : 0092.00GB
Space on drive
        Available space on drive : 0170.05GB
--------------------------------------------

2024/08/12 13:16:15-INFO-DeleteOldObservations-line:197 - AU000A_20240806_100803_187106, size 15.1GB marked fo$
2024/08/12 13:16:16-INFO-DeleteOldObservations-line:212 - Deleted AU000A_20240806_100803_187106
2024/08/12 13:16:16-INFO-DeleteOldObservations-line:596 -

--------------------------------------------
Directory quotas after management
--------------------------------------------
Space used
   Total space used for RMS_data : 0084.28GB
            bz2 files space used : 0001.97GB
 archived directories space used : 0004.52GB
              total for archives : 0006.48GB
   Captured directory space used : 0077.50GB
Quotas allowed
        Total quota for RMS_data : 0100.00GB
                  bz2 file quota : 0003.00GB
      Archived directories quota : 0005.00GB
          Remaining for captured : 0092.00GB
Space on drive
        Available space on drive : 0185.21GB
--------------------------------------------
```

dvida, 2024-08-13:
Good for merging?

g7gpr, 2024-08-13:
Into pre-release.

Kind regards

David Rollinson

I am writing this on a phone, apologies for brevity and errors.

closed 2024-08-13 by g7gpr