CroatianMeteorNetwork/RMS #344: FileNotFoundError exception does not work with Python 2?

Cybis320, 2024-08-11:
Apparently, `FileNotFoundError` exception does not exist in Python 2.7. It should raise a NameError. Has anyone seen this issue ? A fix is to to replace `FileNotFoundError` with `IOError`.  I have not tested yet.

markmac99, 2024-08-11:
Yes, python2's error handling is much more primitive and often you can't filter errors as nicely as you'd like to. I don't think there's a workaround.

dvida, 2024-08-11:
this has been fixed in the recent PR

Cybis320, 2024-08-11:
I wouldn't say it was fixed yet. There are lot of filenotdound exceptions left which I believe would crash of python 2 if raised. 

I'm not setup to test python 2 yet - but if anyone could check if replacing with IOError is good enough on python 2 and 3. It would catch more errors than just file not found, but is that a problem?

markmac99, 2024-08-11:
I agree it should be harmless to replace FileNotFoundError with IOerror. Rather than search and replace in many files a workaround that is used in binviewer is this:
```
try:
    FileNotFoundError
except NameError:
    FileNotFoundError = IOError
```
A slightly neater version would do 
```
if sys.version_info[0] < 3:
    FileNotFoundError = IOError
```

Cybis320, 2024-08-11:
Great idea

markmac99, 2024-08-11:
By the way, i suggest that soon we need to make a call on continuing active support for Python 2, which went End of Life in 2020. More and more libraries are withdrawing support for the old version, and it will get harder, and messier, to support. 

One method would be to create a new "legacy" branch and add something to the RMS_Update script to switch Python2 enviornments to that branch. We could then start using modern python features like f-strings, better error handling, newer versions of libraries and so on, without having to make the requirements file or setup script more and more complex, and without having to maintain antique builds to test on.

dvida, 2024-08-11:
I agree. There aren't that many Pis running Pi3s, but those are mostly in low-income countries. I'd say we try to push it another year and see how it goes. If it becomes hard to support it, we'll announce it and transition to the legacy branch. For now, it has been reasonably trivial to support Python 2. Some parts of the code don't work in Python 2 (e.g. the flux code) so the server picks up the slack.

If you'd like, I could soon announce our plans so people could switch out their Pi3s in the next year.

Cybis320, 2024-08-11:
The codebase currently uses a mix of `FileNotFoundError` (~40%) and `IOError` (~60%). Here’s a quick recap of their differences:

- IOError: Handles both cases where a file is not found and where a file is found but is corrupted or inaccessible for other reasons.
- FileNotFoundError: Specifically handles cases where a file is not found. If a file is found but corrupted, it will later raise an unhandled exception.

Given this, wouldn’t it make sense in most cases to handle both scenarios—missing and corrupted files—under IOError?

Are there specific cases where we would want to distinguish these errors and prefer an unhandled exception for corrupted files, or could we streamline the code by replacing FileNotFoundError with IOError?

Both options are easily implementable. Either:

1. Add the following code at the beginning of every file where FileNotFoundError is used:

```
# Map FileNotFoundError to IOError in Python 2 as it does not exist
if sys.version_info[0] < 3:
    FileNotFoundError = IOError

```

2. Or, replace every instance of FileNotFoundError with IOError directly.

dvida, 2024-08-12:
I endorse option 1. That way the code will stay up to date and require a tiny intervention, with no new bugs introduced when running on machines with Py3.

g7gpr, 2024-08-17:
I bought new Pi3, the latest firmware is not completely compatible with the legacy images.

Cybis320, 2024-08-22:
#346 merged into prerelease. Closing.