CroatianMeteorNetwork/RMS #349: Multiple values for keyword argument 'refraction'

Cybis320, 2024-08-14:
I'm getting this error with prerelease on a MuliCamLinux machine:
```
2024/08/10 12:21:27-INFO-QueuedPool-line:204 - All workers are idle!
Inserting poison pills...
2024/08/10 12:21:27-INFO-QueuedPool-line:204 - Inserting poison pills...
Inserting pill 1
2024/08/10 12:21:27-INFO-QueuedPool-line:204 - Inserting pill 1
Closing pool...
2024/08/10 12:21:28-INFO-QueuedPool-line:204 - Closing pool...
Terminating pool...
2024/08/10 12:21:28-INFO-QueuedPool-line:204 - Terminating pool...
Joining pool...
2024/08/10 12:21:28-INFO-QueuedPool-line:204 - Joining pool...
2024/08/10 12:21:28-INFO-StartCapture-line:480 - Detection finished!
2024/08/10 12:21:28-INFO-StartCapture-line:517 - Collecting results...
2024/08/10 12:21:29-INFO-DetectStarsAndMeteors-line:147 - TOTAL: 30 detected meteors.
2024/08/10 12:21:31-INFO-Reprocess-line:184 - Filtering out detections using machine learning...
2024/08/10 12:21:31-INFO-MLFilter-line:400 - ML filtering starting...
2024/08/10 12:21:31-WARNING-MLFilter-line:431 - The package tflite_runtime is not installed! This package is not available on Python 2.
2024/08/10 12:21:31-WARNING-MLFilter-line:432 - ML filtering skipped...
2024/08/10 12:21:31-INFO-DownloadPlatepar-line:25 - Checking for new platepar on the server...
2024/08/10 12:21:31-DEBUG-DownloadPlatepar-line:30 - Establishing SSH connection to: gmn.uwo.ca:22...
2024/08/10 12:21:31-INFO-UploadManager-line:73 - Trying ssh-agent key b'cc162fc08dd711074df3e7a7d0f9eea9'
2024/08/10 12:21:31-WARNING-UploadManager-line:82 - ... failed! - Authentication failed.
2024/08/10 12:21:31-INFO-Reprocess-line:75 - Loaded platepar from RMS directory: platepar_cmn2010.cal
2024/08/10 12:21:31-INFO-CheckFit-line:529 - Total calstars: 26308
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/bolide/source/RMS/RMS/StartCapture.py", line 1174, in <module>
    night_archive_dir = runCapture(config, duration=duration, nodetect=cml_args.nodetect, \
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bolide/source/RMS/RMS/StartCapture.py", line 530, in runCapture
    night_archive_dir, archive_name, _ = processNight(night_data_dir, config, \
                                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bolide/source/RMS/RMS/Reprocess.py", line 202, in processNight
    platepar, fit_status = autoCheckFit(config, platepar, calstars_list)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bolide/source/RMS/RMS/Astrometry/CheckFit.py", line 549, in autoCheckFit
    n_matched, avg_dist, cost, _ = matchStarsResiduals(config, platepar, catalog_stars, star_dict, \
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bolide/source/RMS/RMS/Astrometry/CheckFit.py", line 83, in matchStarsResiduals
    fov_radius = getFOVSelectionRadius(platepar)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bolide/source/RMS/RMS/Astrometry/ApplyAstrometry.py", line 390, in getFOVSelectionRadius
    _, ra_data, dec_data, _ = xyToRaDecPP(time_data, x_data, y_data, level_data, platepar, \
                              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bolide/source/RMS/RMS/Astrometry/ApplyAstrometry.py", line 630, in xyToRaDecPP
    RA_data, dec_data = cyXYToRADec(JD_data, np.array(X_data, dtype=np.float64), \
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "RMS/Astrometry/CyFunctions.pyx", line 1070, in RMS.Astrometry.CyFunctions.cyXYToRADec
TypeError: cyXYToRADec() got multiple values for keyword argument 'refraction'
```
Anyone else having this issue?

dvida, 2024-08-14:
That's very strange, which branch are you running?

Cybis320, 2024-08-14:
prerelease

Cybis320, 2024-08-14:
The problem is that the refraction parameter is being passed twice: once as part of the platepar object's attributes that are being unpacked, and again as an explicit keyword argument.

dvida, 2024-08-14:
Are you running the latest version of the code? The current line numbers don't match to what your error reports.

Cybis320, 2024-08-14:
Argh nope, I'm not. I forgot to uncomment git pull.

Cybis320, 2024-08-15:
Ok, I'm still seeing the issue on prerelease. Reopening.

Cybis320, 2024-08-17:
Anybody else running prerelease in with Multicam? Are you having the same issue?
If I run a test run with no stars, everything is fine. But after an actual night post-processing cannot complete and terminate with this error (I'm not sure why it says line 1070 in CyFunctions, it might be a compiled cython thing?):
```
2024/08/17 03:35:18-INFO-ApplyRecalibrate-line:688 - MAX -1.8019116466305
2024/08/17 03:35:18-INFO-ApplyRecalibrate-line:415 - Processing: FF_US9992_20240817_030526_760_0000512.fits
2024/08/17 03:35:18-INFO-ApplyRecalibrate-line:416 - ------------------------------------------------------------------------------
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/bolide/source/RMS/RMS/StartCapture.py", line 867, in <module>
    runCapture(config, duration=duration, nodetect=cml_args.nodetect, upload_manager=upload_manager, \
  File "/home/bolide/source/RMS/RMS/StartCapture.py", line 530, in runCapture
    night_archive_dir, archive_name, _ = processNight(night_data_dir, config, \
                                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bolide/source/RMS/RMS/Reprocess.py", line 228, in processNight
    recalibrated_platepars = recalibrateIndividualFFsAndApplyAstrometry(night_data_dir, \
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bolide/source/RMS/RMS/Astrometry/ApplyRecalibrate.py", line 697, in recalibrateIndividualFFsAndApplyAstrometry
    recalibrated_platepars = recalibratePlateparsForFF(
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bolide/source/RMS/RMS/Astrometry/ApplyRecalibrate.py", line 437, in recalibratePlateparsForFF
    result, min_match_radius = recalibrateFF(
                               ^^^^^^^^^^^^^^
  File "/home/bolide/source/RMS/RMS/Astrometry/ApplyRecalibrate.py", line 161, in recalibrateFF
    n_matched, avg_dist, cost, _ = CheckFit.matchStarsResiduals(
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bolide/source/RMS/RMS/Astrometry/CheckFit.py", line 83, in matchStarsResiduals
    fov_radius = getFOVSelectionRadius(platepar)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bolide/source/RMS/RMS/Astrometry/ApplyAstrometry.py", line 433, in getFOVSelectionRadius
    _, ra_data, dec_data, _ = xyToRaDecPP(time_data, x_data, y_data, level_data, platepar, \
                              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bolide/source/RMS/RMS/Astrometry/ApplyAstrometry.py", line 673, in xyToRaDecPP
    RA_data, dec_data = cyXYToRADec(JD_data, np.array(X_data, dtype=np.float64), \
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "RMS/Astrometry/CyFunctions.pyx", line 1070, in RMS.Astrometry.CyFunctions.cyXYToRADec
TypeError: cyXYToRADec() got multiple values for keyword argument 'refraction'
```

Cybis320, 2024-08-17:
Not sure if it matters, but my setup is currently not running ML due to it not being supported with python 3.12 without the full tensorflow PR.

Cybis320, 2024-08-21:
Solved by cleaning up *.so files. PR #371. Closing.