CroatianMeteorNetwork/RMS #351: CryptographyDeprecationWarning: Blowfish

Cybis320, 2024-08-15:
I was wondering why we are restricting paramiko in such a way:

```python
paramiko==2.4.0,<=2.8.1; python_version=='2.7'
paramiko>=2.4.2,<=2.8.1; python_version>='3.6'
```

Paramiko 2.12.0 supports Python 2.7. Are recent paramiko versions not working for us somehow? I've installed paramiko 3.4.1, and nothing seems to break, although I haven't tried transferring files. The nice thing is that it gets rid of the deprecation warning. 

Should we find out what minimum version suppresses the deprecation warning and update the requirements accordingly? Perhaps something like:

```python
paramiko>=2.4.0,<=2.12.0; python_version=='2.7'
paramiko>=3.x.x; python_version>='3.6'
```

dvida, 2024-08-15:
Agreed, newer versions on python 3 should be tested. There was an issue that was breaking file transfers, so we degraded it. This was a few years ago.

Cybis320, 2024-08-15:
Looks like version >=2.9.0 removed the use of Blowfish and other older ciphers by default. Changing our requirements to this version will effectively update all current installations to the most recent Paramiko version compatible with the installed version of Python:
 
1. Our current requirements are:
   ```
   paramiko==2.4.0,<=2.8.1; python_version=='2.7'
   paramiko>=2.4.2,<=2.8.1; python_version>='3.6'
   ```
   This caps Paramiko at version 2.8.1 for both Python 2 and 3.

2. By changing to `paramiko>=2.9.0`, we're specifying a minimum version higher than our current maximum. This will trigger updates on all systems when RMS_update runs. 

3. The update outcomes will be:
   - Python 2 environments will install Paramiko 2.12.0 (the last version supporting Python 2.7)
   - Python 3 environments will install Paramiko 3.4.1 (the latest version as of now)

4. It eliminates the need to differentiate between Python 2 and 3 in our requirements.

5. It resolves the Blowfish deprecation warning across all our systems.

6. It makes the code more secure. 

Proposed updated requirement:
```
paramiko>=2.9.0
```

But obviously we should test that it doesn't affect file transfers negatively.

markmac99, 2024-08-22:
As noted on the PR, paramiko>=2.9.0  fails to install on Python 2.7 / Jessie, which is the last Pi3 build we support as unfortunately cryptography 3.3.2, the terminal version that supports Python2.7, fails to build with an SSL compatability issue. Its possible we can patch openSSL but thats a lot of risky work. Simpler to keep python 2.7 's current restrictions. 

Next I will test a vanilla buster-32 build on Pi4, which is probably one of the most common amongst our stations. This runs python 3.7.

markmac99, 2024-08-22:
Looks ok on Buster32, Python 3.7, Pi4. I was able to upload dummy data to my sftp server with RMS/UploadManager.py 
```
(vRMS) pi@testpi4:~/source/RMS $ python -m RMS.UploadManager
Failure whilst checking that directory files/upload_test/dir2/dir3 exists
Directory 'files' already exists.
Directory 'files/upload_test' created.
Directory 'files/upload_test/dir2' created.
Directory 'files/upload_test/dir2/dir3' created.
2024/08/22 22:38:59-INFO-UploadManager-line:219 - Establishing SSH connection to: ukmonhelper2:22...
2024/08/22 22:38:59-INFO-UploadManager-line:73 - Trying ssh-agent key b'50997d3a2656bfad9129d3f6ec626ef7'
2024/08/22 22:38:59-INFO-UploadManager-line:78 - ... success!
2024/08/22 22:38:59-INFO-UploadManager-line:273 - Copying /tmp/test.txt (0.00MB) to files/test.txt
2024/08/22 22:38:59-INFO-UploadManager-line:287 - File upload verified: files/test.txt
2024/08/22 22:38:59-INFO-UploadManager-line:492 - Upload successful!

```
Note that paramiko and pynacl both get updated. 
```
Successfully installed paramiko-3.4.1 pynacl-1.5.0
```

dvida, 2024-08-23:
Hi Mark,
Is the suggestion then to leave things as they are for the Py2 (which we'll officially deprecate some time next year), and only update the Py3 version?

markmac99, 2024-08-23:
Yes, thats correctl Luc has made the necessary change already in the PR i think. 

I also tested with Bookworm64/Python3.9/Pi4 and all worked fine so i think this is good to go

Cybis320, 2024-08-23:
Merged. Sticking with 2.4.0 for py 2. Install latest for py 3. Closing.