CroatianMeteorNetwork/RMS #359: readme.md - broken link

nzstevem, 2024-08-17:
In reademe.md this link is broken. 

" ... is called Kernel-based Hough Transform, and you can read more about it here: [KHT](http://www2.ic.uff.br/~laffernandes/projects/kht/)"

Also, are the ip camera connection strings correct ? In particular are the pieces past the first "!" symbol required?

This is what is suggested for a raspi:

device: rtspsrc location=rtsp://192.168.42.10:554/user=admin&password=&channel=1&stream=0.sdp ! rtph264depay ! queue ! h264parse ! omxh264dec ! queue ! videoconvert ! appsink sync=1

In practice nothing after and including the first ! is necessary (or is present in the std image).

Cybis320, 2024-08-17:
It's outdated. The current recommended device url is:
```
rtsp://192.168.42.10:554/user=admin&password=&channel=1&stream=0.sdp
```
It looks like the whole doc could be updated.

satmonkey, 2024-08-17:
IMHO, for the Pi3, the OMX decoder is still needed, therefore the
pipeline is HW dependent.