CroatianMeteorNetwork/RMS #364: Bug in plotting timestamp intervals

markmac99, 2024-08-18:
The master branch fails during Reprocess with this error
```
024/08/18 11:15:34-INFO-Reprocess-line:396 - Plotting timestamp intervals...
Tar file not found.
2024/08/18 11:15:34-DEBUG-Reprocess-line:408 - Plotting timestamp interval failed with message:
TypeError('unsupported format string passed to NoneType.__format__')
2024/08/18 11:15:34-DEBUG-Reprocess-line:409 - ['Traceback (most recent call last):\n', '  File "/home/pi/source/RMS/RMS/Reprocess.py", line 402, in processNight\n    jitter_quality, dropped_frame_rate))\n', 'TypeError: unsupported format string passed to NoneType.__format__\n']
2024/08/18 11:15:34-INFO-Reprocess-line:525 - Archiving detections to /home/pi/RMS_data/ArchivedFiles/UK001L_20240817_200017_648946
```
the code is enclosed in a try/except so it does not halt RMS but its presumably not creating some graph.  I'll investigate it.

markmac99, 2024-08-18:
ok, its a simple problem: if the Fieldsums bz2 archive isn't available, then plotFFTimeIntervals() returns None,None,None. However the values are then passed to log.info, which can't print None as a float.
The fix is to check for None before printing. I'll create a fix branch.

Cybis320, 2024-08-18:
I think this is fixed in prerelease, no?