CroatianMeteorNetwork/RMS #369: ObservationSummary: UnboundLocalError: cannot access local variable 'pp'

Cybis320, 2024-08-18:
I'm getting this error on MCL:
```
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/bolide/source/RMS/RMS/StartCapture.py", line 873, in <module>
    runCapture(config, duration=duration, nodetect=cml_args.nodetect, upload_manager=upload_manager, \
  File "/home/bolide/source/RMS/RMS/StartCapture.py", line 536, in runCapture
    night_archive_dir, archive_name, _ = processNight(night_data_dir, config, \
                                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bolide/source/RMS/RMS/Reprocess.py", line 580, in processNight
    observation_summary_path_file_name, observation_summary_json_path_file_name = finalizeObservationSummary(config, night_data_dir, pp)
                                                                                                                                     ^^
UnboundLocalError: cannot access local variable 'pp' where it is not associated with a value
```

Cybis320, 2024-08-18:
pp could be unassigned before being called because it's defined within an if-block:
```python
    # If FFs are not uploaded, choose two to upload
    if config.upload_mode > 1:
    
        # If all FF files are not uploaded, add two FF files which were successfully recalibrated
        recalibrated_ffs = []
        if recalibrated_platepars is not None:
            for ff_name in recalibrated_platepars:

                pp = recalibrated_platepars[ff_name]

                # Check if the FF was recalibrated
                if pp.auto_recalibrated:
                    recalibrated_ffs.append(os.path.join(night_data_dir, ff_name))
```
then later:
```python
    observation_summary_path_file_name, observation_summary_json_path_file_name = finalizeObservationSummary(config, night_data_dir, pp)
```

g7gpr, 2024-08-18:
Refer pr #361

closed 2024-08-21 by Cybis320