CroatianMeteorNetwork/RMS #370: MaskImage - NoneType has no attribute 'shape'

g7gpr, 2024-08-18:
```
2024/08/19 06:29:58-DEBUG-Reprocess-line:328 - 

['Traceback (most recent call last):\n', 
' File "/home/au0028/source/RMS/RMS/Reprocess.py", line 324, in processNight
\n    prepareFluxFiles(config, night_data_dir, os.path.join(night_data_dir, ftpdetectinfo_name))\n', 
'  File "/home/au0028/source/RMS/Utils/Flux.py", line 3686, in prepareFluxFiles\n    
getCollectingArea(dir_path, config, flux_config, platepar, mask)\n',
 '  File "/home/au0028/source/RMS/Utils/Flux.py", line 1851, in getCollectingArea\n    
col_areas_ht = collectingArea(platepar, mask=mask, side_points=flux_config.side_points, \\\n                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^\n', 
'  File "/home/au0028/source/RMS/Utils/Flux.py", line 1605, in collectingArea\n    
mask = MaskStructure(None)\n           ^^^^^^^^^^^^^^^^^^^\n', 
'  File "/home/au0028/source/RMS/RMS/Routines/MaskImage.py", line 40, in __init__\n    self.height = img.shape[0]\n                  ^^^^^^^^^\n', 
"AttributeError: 'NoneType' object has no attribute 'shape'\n"]
```

I've not investigated this in any detail yet. Please let me know further information is required.

Cybis320, 2024-08-21:
I see the same message.

dvida, 2024-08-22:
Can someone track git blame to figure out the change that's breaking this? I don't think we recently did anything in the mask functions.

Cybis320, 2024-08-22:
there was a change to MaskStructure last week:
```python
class MaskStructure(object):
    def __init__(self, img):
        """ Structure for holding the mask. This is used so the mask can be hashed. """
        
        self.img = img

        # Get the mask resolution
        self.height = img.shape[0]
        self.width = img.shape[1]
```

Cybis320, 2024-08-22:
it probably needs to check that img exist first:
```python
def __init__(self, img):
    """ Structure for holding the mask. This is used so the mask can be hashed. """
    self.img = img
    # Get the mask resolution
    if img is not None:
        self.height = img.shape[0]
        self.width = img.shape[1]
    else:
        self.height = None
        self.width = None
```

dvida, 2024-08-22:
That's probably it! Could you please open a PR?

Cybis320, 2024-08-22:
I'm wondering if it's just leftover debugging code. The lines are actually not doing anything as far as I can tell. I'll submit a PR with deletions.

dvida, 2024-08-22:
Fixed in the latest commit.

closed 2024-08-22 by dvida