CroatianMeteorNetwork/RMS #372: Compare configs unicode encode error

g7gpr, 2024-08-20:
Observed in logs on a Pi5 running standard distribution.

``
2024/08/20 22:27:36-DEBUG-Reprocess-line:464 - ['Traceback (most recent call last):\n', 
'  File "/home/rms/source/RMS/RMS/Reprocess.py", line 456, in processNight\n    
                   f.write(compareConfigs(config_file_path,\n', "UnicodeEncodeError: 'latin-1' codec can't encode character '\\u2022' in position 570: ordinal not in range(256)\n"]
```

Results in an empty file

```
-rw-r--r-- 1 rms    0 Aug 20 22:27 AU0004_20240820_101725_619305_config_audit_report.txt
```

Cybis320, 2024-08-23:
Is this solved with the #360 merge?

g7gpr, 2024-08-24:
Last seen on 20240822, so yes.

closed 2024-08-24 by g7gpr