CroatianMeteorNetwork/RMS #376: No mask found even though mask is present

Cybis320, 2024-08-21:
On prerelease Multicam, with `mask.bmp` located in `~/source/Stations/[Station_ID]/` I get:
```
2024/08/21 02:43:30-INFO-DetectionTools-line:69 - No mask file has been found.
```

Cybis320, 2024-08-21:
@g7gpr, is there a change in the required folder structure? I'm also getting:
```
Starting RMS...
Using config from /home/bolide/source/Stations/US9992/.config
/home/bolide/source/Stations/US9992/.config
Disabled upload because the default station code is used!
```
The `.config` in the RMS root folder has the default station ID but the `.config` in `~/source/Stations/[Station_ID]/` is correct.

Cybis320, 2024-08-22:
In StartCapture, it's loading the config from the working dir, which is set to ~/source/RMS by the new multicam StartCapture script:
```python
# Load the config file
config = cr.loadConfigFromDirectory(cml_args.config, os.path.abspath('.'))
```

Cybis320, 2024-08-22:
The same issue is present in multiple locations: BufferedCapture, DownloadMask, EventMonitor, and StartCapture

Cybis320, 2024-08-22:
Also, ConfigReader doesn't log, it just print to the terminal, so the issue is not visible in the log.

g7gpr, 2024-08-22:
Are you passing the station name as an argument to StartCapture?

Cybis320, 2024-08-22:
Yes.

g7gpr, 2024-08-22:
I think what you are seeing is the config reader initialization where it reads the default config from . 

https://github.com/CroatianMeteorNetwork/RMS/blob/6f42cc1fd2f40b65b95e6d20a3724dcb8b900835/RMS/ConfigReader.py#L235

But this gets overwritten by the config specified on the command line. 

As for the mask, this is very similar to the error that is raised when a mask of the wrong format is used; for example the mask might have been saved with color information, rather than with the "Do not write color space option selected"

In any case, I can't replicate this issue. However, I'm happy to submit a pull request to revert the change if needed.

Cybis320, 2024-08-22:
Dave, you are right! Thank you! I connected the wrong dots here. Apologies for the extra work.

The issue actually was not that the mask.bmp was in the wrong format. It turns out 'No mask file has been found' is outputted even if a mask is found but is all white. So it's all good!

I might post a PR to make the log entry less confusing.

Regarding the default station message: The 'Disabled upload' message is only triggered when an actual .config with XX0001 is loaded. When I change the root .config stationID to something other than XX0001, the warnings disappear. It looks like the four *.pyxbld files are loading the root .config:
https://github.com/CroatianMeteorNetwork/RMS/blob/6f42cc1fd2f40b65b95e6d20a3724dcb8b900835/RMS/CompressionCy.pyxbld#L9-L16
I don't know how big of a deal that is. My assumption was that on a Multicam setup, the root config should be ignored. However, the extra_compile_args are read from the root config.

g7gpr, 2024-08-22:
The init of the config reads from that location. Hard coded I think, but not an issue. 

https://github.com/CroatianMeteorNetwork/RMS/blob/3313489fffe1c9b6148a85cb205a4ffe37575be5/RMS/ConfigReader.py#L225

Cybis320, 2024-08-22:
> The init of the config reads from that location. Hard coded I think, but not an issue.
> 
> https://github.com/CroatianMeteorNetwork/RMS/blob/3313489fffe1c9b6148a85cb205a4ffe37575be5/RMS/ConfigReader.py#L225

Right, but that's not what's triggering the multiple 'disabled updload' messages - the pyxbld files are the triggers by actually loading the config in the working directory (~/source/RMS) with potentially the default stationID. I guess I'm not understanding the correlation between line 225 and the messages. But I do understand the messages are just spurious and the uploads are not actually disabled as the correct config is eventually loaded.

However, the extra compile arguments are coming from the root config and not the stations config - which could be the desired behavior anyway as each stations can't have different compile arguments? It just seems a little unclear what the desire behavior should be. Maybe, it's just a matter of documenting somewhere that the root config is not entirely ignored in MCL. Maybe I can add something is the .config comments - what do you think?

Also, and a bit unrelated, we import the whole RMS just to get the `__file__`, should just import the `__file__` instead? :
Currently:
https://github.com/CroatianMeteorNetwork/RMS/blob/58cc62c09211f896fcde98097532650d93738dba/RMS/ConfigReader.py#L23

Change to this? Not sure if it would have an impact on memory usage.
```python
from RMS import __file__ as RMS_file
self.rms_root_dir = os.path.abspath(os.path.join(os.path.dirname(RMS_file), os.pardir))
```

g7gpr, 2024-08-22:
I don't really have a view.

closed 2025-05-31 by Cybis320