CroatianMeteorNetwork/RMS #380: Flaw in logic in latest TrackStack release

markmac99, 2024-08-22:
I should have spotted this when code-reviewing it:

the textoption optional argument is defaulted to 0, which seems reasonable:
```
def trackStack(dir_paths, config, border=5, background_compensation=True, 
...
        textoption=0):
```
HOWEVER if passed in as a cmdline argument it is a list of ints, default [0], and so the code uses it as such eg at line 347.
```
if textoption[0] == 1:
```
I didn't spot this because i only tested it as a `commandline` tool, i didn't test it when called from Python, when the default 0 will cause a crash:
```
    if textoption[0] == 1:
TypeError: 'int' object is not subscriptable
``` 
I'll create a patch release and pull request today.

g7gpr, 2024-08-22:
Thanks, that was my first interaction with cml.args. Better check the rest of my work.

markmac99, 2024-08-22:
ROFL! I should have spotted it, and i thought i was so clever with my chaos engineering tests... missed the one crucial one.  

Have retested with t=0, 1, 2, -1, 99999, and potato and just testing calling it directly from python. All seems ok now but its veeerrrrry slow on the production branch !

markmac99, 2024-08-22:
Tested successfully.

g7gpr, 2024-08-23:
Tested successfully also. However, I subimtted a PR to this branch to fix the docstring.

```
import os.path
from RMS.Misc import mkdirP
import RMS.ConfigReader as cr

from TrackStack import trackStack

if __name__ == "__main__":
    # ## PARSE INPUT ARGUMENTS ###
    # Init the command line arguments parser

    file_path = os.path.expanduser(("~/tmp/AU0028_20240814_123650_883664/"))

    config = cr.loadConfigFromDirectory(".config", file_path)

    test_list = [0,1,2,-1,'text_option',65536]
    result_dir = os.path.expanduser("~/tmp/trackstacktesting")

    for text_option, dest_folder in zip(test_list, range(0,len(test_list))):

        stack_out_dir = os.path.join(result_dir,str(dest_folder))
        mkdirP(result_dir)
        print("Writing to {:s}".format(result_dir))
        trackStack([file_path], config, background_compensation=False,
            hide_plot=False, showers=None,
            darkbackground=False, out_dir=result_dir, scalefactor=2,
            draw_constellations=True, one_core_free=True,
            textoption=text_option)
```

markmac99, 2024-08-23:
patched in your docstring update into the PR

closed 2024-08-23 by markmac99