CroatianMeteorNetwork/RMS #387: Current Config Audit Results on Prerelease

Cybis320, 2024-08-25:
FYI, there's quite the discrepancy between the current `.config` and ConfigReader implemented options. The current omit list (options silenced in the audit):
```python
OMIT_FROM_CONFIG = {'lat',
                    'lon',
                    'location',
                    'altitude',
                    'event_monitor_db_name',
                    'force_v4l2',
                    'mask',
                    'platepar_name'
                    }

```

The current audit results:
```
================================================================================
                         DEV CONFIG COMPARISON RESULTS                          
================================================================================

    NOTE: .configTemplate file not found. Some comparisons will be skipped.     

               OPTIONS NOT IN CONFIG FILE BUT IMPLEMENTED IN RMS:               
--------------------------------------------------------------------------------
 - arch_dir_quota
 - brightness
 - bz2_files_quota
 - capt_dirs_to_keep
 - contrast
 - dark_file
 - mask_download_permissive
 - mask_remote_name
 - min_lines
 - platepar_template_dir
 - quota_management_disabled
 - remote_mask_dir
 - rms_data_quota
 - use_dark

                     OPTIONS COMMENTED OUT IN .CONFIG FILE                      
                          Default values will be used:                          
--------------------------------------------------------------------------------
 - public_elevation
 - public_latitude
 - public_longitude
 - star_catalog_band_ratios

================================================================================
                             INCOMPLETE COMPARISON                              
       Some files were not found, resulting in an incomplete comparison.        
================================================================================

```

Do we want to silence some of these options, or add them to config?
-Luc

Cybis320, 2024-08-27:
1. #### Add to OMIT list as it seems deprecated?:

 - contrast
 - brightness
 

2. #### Reopen and merge PR #352?:

 - capt_dirs_to_keep
 - quota_management_disabled
 - rms_data_quota
 - arch_dir_quota
 - bz2_files_quota
 
3. #### Accidentally deleted by commit [#79b061](https://github.com/CroatianMeteorNetwork/RMS/commit/79b0617952482e6880acf8713917cccb7d6d7d4b) ? Add back in config?:
 - dark_file
 - use_dark

4. #### Correct Configreader and add to config?:
- min_lines

In addition of not being in config, there seems to be an error in ConfigReader on this one:

https://github.com/CroatianMeteorNetwork/RMS/blob/1381e4bfbdd4dda998f7e193117a10d541f67f32/RMS/ConfigReader.py#L1332-L1333
Since min_lines does not exist max_lines option is ignored.

5. #### PR #239 introduced these without associated config options. Add them to config?:
 - mask_download_permissive
 - mask_remote_name
  - remote_mask_dir

6. #### Commit https://github.com/CroatianMeteorNetwork/RMS/commit/03ec6b6c59a05b0fc7c1123dcdd128a1cbace6d3 introduced this one. Not sure if it should be added to config or omit:
 - platepar_template_dir

g7gpr, 2024-08-31:
Q5, I suggest to add mask_download_permissive to .config, set to True, and remove the other two options and hard code them to prevent unintentional changes. No other comments from me.

Cybis320, 2025-05-31:
solved

closed 2025-05-31 by Cybis320