CroatianMeteorNetwork/RMS #389: Preview Observation summary - output location, camid

nzstevem, 2024-08-26:
Looking at preview code wondering if txt file output location rms_data/ is going to work. Would it not be better if the output went back to the nightly capture directory so they are preserved day to day, and potentially Included in uploads.

Also thinking about multi-cam configurations. Location will need to be camera specific.

I think including the camid in the attributes would be very useful particularly as the output filename doesn’t include this.

g7gpr, 2024-08-27:
I think it does go to the Captured and Archive directories. Can you paste a link to the code that makes you think otherwise?

g7gpr, 2024-08-27:
The filename does include the station, but adding into the json and text file seems useful.

nzstevem, 2024-08-27:
I can't find the code I was worried about again. From memory it was in the  *** if __name__ == "__main__": *** section at the bottom. There was a line included the full "...observation_summary.txt" file name.  No matter. Looks good.

g7gpr, 2024-08-27:
Thanks, I'll work on adding the stationID.

closed 2024-09-01 by nzstevem