CroatianMeteorNetwork/RMS #398: Exception handling in Observation Summary

g7gpr, 2024-08-31:
Has this symbol been defined? 

https://github.com/CroatianMeteorNetwork/RMS/blame/441008b356fa203df26d8c5094c82797de5cf7bf/RMS/Formats/ObservationSummary.py#L140

The issue could well be my lack of understanding of Python exception handling; if someone can confirm it is OK, then I'll close the issue.

Cybis320, 2024-09-01:
I apologize, I don't remember making this commit or the rational behind it even though clearly I did. It's missing some imports:
```python
from git import Repo, InvalidGitRepositoryError
import os

def get_rms_repo():
    rms_path = os.path.expanduser("~/source/RMS")
    try:
        return Repo(rms_path)
    except InvalidGitRepositoryError:
        print(f"Warning: No valid Git repository found at {rms_path}")
        return None
```
Do you want me to submit a PR? Sorry for messing up your code...

g7gpr, 2024-09-01:
Probably best you submit a PR, as I'd be committing code I don't understand. Thanks for getting back to me on it.

Cybis320, 2024-09-01:
It got back to me. I was concerned about searching for the repo in the parent directory when it's not guaranteed the repo is located there. In the meantime I've added PR #391. When/if that PR is merged, getRmsRootDir() should be the most reliable way to get the path to the repo. So, I think the best course of action is to revert my commit and wait for #391. Once merged, change startObservationSummaryReport from:
```python
repo = git.Repo(search_parent_directories=True)
```

To:
```python
rms_root = getRmsRootDir()
repo = git.Repo(rms_root)
 ```

I'll submit a PR to revert my commit

g7gpr, 2024-09-06:
Resolved, thank you.