CroatianMeteorNetwork/RMS #406: File handle gets created, but never destroyed

g7gpr, 2024-09-03:
https://github.com/CroatianMeteorNetwork/RMS/blob/33bc410719930367f3e85301f84a9dfc14971478/RMS/Formats/FFbin.py#L46

Apologies if this is my poor understanding of the code, but it seems that the file handle here gets created, but never destroyed. It does not seem to be using a context handler. I'm not sure how to fix this, because it is often used within a try block, and the handler does not get passed to the except clause.

g7gpr, 2024-09-03:
Can you give a second opinion on this @Cybis320?

g7gpr, 2024-09-03:
https://github.com/CroatianMeteorNetwork/RMS/blob/33bc410719930367f3e85301f84a9dfc14971478/RMS/Formats/FFfits.py#L77

Maybe a similar issue here.

g7gpr, 2024-09-03:
https://github.com/astropy/astropy/issues/7404

Describes the issue in detail. Astropy holds the HDU in a memory map, which seems to keep a different file handle open. Passing 

``
memmap=False
```
solves the issue. Fixing this is quite a big task, I don't think it affects the normal use of this function, however, it might explain a few things that happen when RMS runs for several days at a time.

g7gpr, 2024-09-03:
With does not fix the problem when opening a FITS file, though it might in the code sample above. Preventing the HDU handler building a memory map does fix the problem.

I discovered this when I tried to create the plot shown below, which opened up every single fits file on a station which contained the star Antares, although it took a bit of head scratching to work out what was going on. 

[image]

Cybis320, 2024-09-03:
Ok, I'm starting to understand I think. The problem here is twofold:

1. The initial file opening with open() is not properly managed.
2. The fits.open() function, which uses memory mapping under the hood, is not properly managed either.

```python

import gc

def read(directory, filename, array=False, full_filename=False):
   
    if (filename.startswith('FF') and ('.fits' in filename)) or full_filename:
        file_path = os.path.join(directory, filename)
    else:
        file_path = os.path.join(directory, "FF_" + filename + ".fits")

    ff = FFStruct()
    hdulist = None

    try:
        hdulist = fits.open(file_path, memmap=True)
        
        # Read the header
        head = hdulist[0].header

        # Read in the data from the header
        ff.nrows = head['NROWS']
        ff.ncols = head['NCOLS']
        ff.nbits = head['NBITS']
        ff.nframes = head['NFRAMES']
        ff.first = head['FIRST']
        ff.camno = head['CAMNO']
        ff.fps = head['FPS']

        # Check for the DATE-OBS field and read datetime from filename if it doesn't exist
        if 'DATE-OBS' in head:
            ff.starttime = head['DATE-OBS']
        else:
            ff.starttime = filenameToDatetimeStr(filename, iso8601=True)

        # Read in the image data
        ff.maxpixel = hdulist[1].data.copy()
        ff.maxframe = hdulist[2].data.copy()
        ff.avepixel = hdulist[3].data.copy()
        ff.stdpixel = hdulist[4].data.copy()

        if array:
            ff.array = np.dstack([ff.maxpixel, ff.maxframe, ff.avepixel, ff.stdpixel])
            ff.array = np.swapaxes(ff.array, 0, 1)
            ff.array = np.swapaxes(ff.array, 0, 2)

    except Exception as e:
        print(f"Error reading FITS file: {e}")
        return None
    finally:
        if hdulist is not None:
            try:
                hdulist.close()
            except Exception as e:
                print(f"Error closing FITS file: {e}")
            
            # Force garbage collection to release memmapped arrays
            for hdu in hdulist:
                if hasattr(hdu, 'data') and hasattr(hdu.data, 'base') and isinstance(hdu.data.base, np.memmap):
                    del hdu.data
            del hdulist
            gc.collect()

    return ff
```

This wrapped the fits.open in 'with' and called garbage collection if there is anything left standing after running out of scope.

g7gpr, 2024-09-03:
Calling fits with memmap = False might be a lower impact fix, but ^ looks splendid.

Incidently, my original comment that the fh does not get closed might be true, but does not seem to be the cause of the problem.

Cybis320, 2024-09-04:
I agree that `memmap = False` might be simpler but I'm not sure what it would do to performance. I didn't test the proposed solution, have you had a chance to?
I agree too that wrapping all `open` in with-blocks is the recommended approach but the current code doesn't cause issues, the fix is very low priority.

g7gpr, 2024-09-04:
I'll test adding a memmap=True default parameter onto read. For my purposes I'll call it as false, and also test what impact setting memmap=False has. Your solution is better, but it's such a big change to such a core piece of the code. I haven't got the guts for that.

g7gpr, 2024-09-04:
Since this has become more of a discussion than a clear statement of a problem, I'll close this and reopen a nice neat one.

closed 2024-09-04 by g7gpr