CroatianMeteorNetwork/RMS #407: angularSeparation uses radians not degrees

g7gpr, 2024-09-04:
https://github.com/CroatianMeteorNetwork/RMS/blob/dcb6f2a3dd617584d93fc2a34dc7036b366e6fdc/RMS/Math.py#L37

Most of RMS uses degrees, however angularSeparation uses radians. Whilst the conversion is trivial, I'd like to move the code 

https://github.com/g7gpr/RMS/blob/e3ea10c1da31e61e6bd172b3d25e8f41beb144ce/RMS/Astrometry/VariableStar.py#L96

into the math library. It wraps angularSeparation in some code that converts to radians, then back to degrees. 

Alternatively I could add an optional pararmeter degrees=False to the existing function, but I don't want to touch existing code; even if nothing could possibly go wrong.

Grateful for any views from @dvida

dvida, 2024-09-05:
Agreed, all functions should use uniform units. It's radians in wmpl and degress in RMS. The best way to do it would be to rework all instances across RMS. I'm also reluctant to touch working code, but it's better than introducing unnecessary complexity. Great care should be taken...

g7gpr, 2024-09-06:
No intention to touch working code, I'll just add a new def into RMS.math, and use it for new development work.

closed 2024-09-06 by g7gpr