CroatianMeteorNetwork/RMS #417: Error when using a video file input for RMS.StartCapture

MaadhyamRana, 2024-09-11:
The exact command used for this is:

`python -m RMS.StartCapture -i ~/Downloads/CAEUA3_20240713_052714_473898_video.mkv`

This was done on the prerelease branch, when testing on a Linux PC.

The error is given by RMS/Formats/ObservationSummary.py, in function startObservationSummaryReport (line 189). It seems to arise from the fact that the duration and input (-d and -i flags respectively) are mutually exclusive arguments. As a result, when an input file is given, the duration is automatically set to None.

The log, along with the exception is as follows (config audit has been left out):

```
Disabled upload because the default station code is used!
Loading the default config!
Creating directory: /home/mrana/RMS_data
   Success: True
Creating directory: /home/mrana/RMS_data/logs
   Sucess: True
2024/09/11 10:58:25-INFO-StartCapture-line:796 - Program start
2024/09/11 10:58:25-INFO-StartCapture-line:797 - Station code: XX0001
2024/09/11 10:58:25-INFO-StartCapture-line:811 - Program version: 20240910_094253, 6b930359ece0a9e6e9fc2c0291362b8fd20b2eef
2024/09/11 10:58:25-INFO-DownloadMask-line:44 - Mask download enabled
2024/09/11 10:58:25-INFO-DownloadMask-line:50 - Checking for new mask on the server...
2024/09/11 10:58:25-DEBUG-DownloadMask-line:55 - Establishing SSH connection to: gmn.uwo.ca:22...
2024/09/11 10:58:25-ERROR-DownloadMask-line:71 - Connecting to server failed!
2024/09/11 10:58:25-INFO-StartCapture-line:892 - Video source: /home/mrana/Downloads/CAEUA3_20240713_052714_473898.mkv

2024/09/11 10:58:25-INFO-DownloadMask-line:44 - Mask download enabled
2024/09/11 10:58:25-INFO-DownloadMask-line:50 - Checking for new mask on the server...
2024/09/11 10:58:25-DEBUG-DownloadMask-line:55 - Establishing SSH connection to: gmn.uwo.ca:22...
2024/09/11 10:58:25-ERROR-DownloadMask-line:71 - Connecting to server failed!
2024/09/11 10:58:25-INFO-DownloadPlatepar-line:24 - Checking for new platepar on the server...
2024/09/11 10:58:25-DEBUG-DownloadPlatepar-line:29 - Establishing SSH connection to: gmn.uwo.ca:22...
2024/09/11 10:58:25-ERROR-DownloadPlatepar-line:45 - Connecting to server failed!
2024/09/11 10:58:25-INFO-Reprocess-line:93 - No platepar file found!
2024/09/11 10:58:25-INFO-StartCapture-line:273 - Initializing frame buffers...
2024/09/11 10:58:26-INFO-StartCapture-line:297 - Initializing frame buffers done!
Loaded 0 backed up results...
2024/09/11 10:58:26-INFO-QueuedPool-line:204 - Loaded 0 backed up results...
Using 1 cores
2024/09/11 10:58:26-INFO-QueuedPool-line:204 - Using 1 cores
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/mrana/RMS/RMS/StartCapture.py", line 895, in <module>
    runCapture(config, duration=None, video_file=video_file, nodetect=cml_args.nodetect,
  File "/home/mrana/RMS/RMS/StartCapture.py", line 386, in runCapture
    log.info(startObservationSummaryReport(config, duration, force_delete=False))
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/mrana/RMS/RMS/Formats/ObservationSummary.py", line 189, in startObservationSummaryReport
    fits_files_from_duration = duration * fps / no_of_frames_per_fits_file
                               ~~~~~~~~~^~~~~
TypeError: unsupported operand type(s) for *: 'NoneType' and 'float'
```

dvida, 2024-09-11:
I belive you were able to fix this today, could you push a PR?

dvida, 2024-09-12:
Yes, make this a separate issue, then try to look in why this is happening.

PR #420 (2024-09-12): Disable observation summary report creation when given video file
Updates RMS/StartCapture.py to not produce an observation summary when using a video file (Resolves #417 )
closed 2024-09-12 by MaadhyamRana