CroatianMeteorNetwork/RMS #418: Incomplete cleanup after running reprocessing job?

peschman, 2024-09-11:
Platform: 8GB Pi5 running Bookworm 64-bit, flashed from RMS_RPi5Bookworm_image_20240910.img.xz, 11-Sep-2024 16:29, 3GB, our newest image as of today and soon to be linked in the wiki. The new image has OS and RMS updated, chrony installed, and the latest default .config file.

I was bringing up a test station last night, and forgot to copy in the mask and platepar file for that camera.

I realized my error in the evening during capture, so in the morning, after processing was completed I moved the ArchivedFiles directory and tar.bz2 out of Archived files, copied in the correct mask and platepar to ~/source/RMS and to the CapturedFiles directory for last night.

I then restarted capture, expecting that the system would detect the missing ArchivedFiles directory and reprocess the data.

It did not detect the missing ArchivedFiles directory and re-create it, so I ran a manual reprocessing job, which completed successfully, using the correct platepar and mask.

I then restarted capture so it would be ready to run tonight, and to my surprise, the system detected unprocessed data, and is now proceeding to duplicate my reprocessing job.

My assumption is that something did not get cleaned up at the end of my reprocessing job that fooled RMS into thinking it needed to reprocess. I believe this problem should be easy to reproduce, as I have a distant recollection of having the same thing happen several months ago.

g7gpr, 2024-09-16:
One of the checks to start reprocessing is if the platepar in the captured directory is newer than the newest FTP file.  I think the sequence of events was

1. Capture with wrong mask and platepar
2. Copy correct mask and platepar into source/RMS
3. Manually reprocess
4. Copy correct mask and platepar into CapturedFiles/Some_capture_Directory
5. System detects that platepar is newer than latest FTP file
6. Automatically reprocesses

I'd be grateful if you can test again, by  swapping steps 4 and 3. This should prevent the automatic reprocess being triggered. 

Thank you.

peschman, 2024-09-17:
I need to clarify the context for the problem
1. flash a  new microSD, copy over .config and mask but forget platepar
2. run first night capture, realize platepar is missing after morning processing
3. copy in platepar to both ~/source/RMS and the previous night's CapturedFiles dir
4. delete ArchivedFiles dir and tar.bz2 from previous night, thinking this will force RMS to reprocesses previous night.
5. but RMS does not reprocess after reboot, so run reprocessing job manually on CapturedFiles dir
6. reprocessing works fine, so reboot 
7. surprise! now RMS reprocesses the data, so it now has been done twice!

It seems that the manual reprocessing job did not correctly finish in such a way that RMS thought that all was good.
I believe this is a repeatable error, as long as the same steps are followed:
- capture without platepar
- copy platepar into both ~/source/RMS and CapturedFiles dir
- manually reprocess, reboot
- watch RMS reprocess again

g7gpr, 2024-09-17:
Looking at your logs, definitely something strange is happening. Thanks for providing the exact sequence required to repeat the testing. At the moment I have no idea why it is happening, but I'll look at it during the next week.