CroatianMeteorNetwork/RMS #432: Deprecation warning in imreg_dft, will become fatal with pip 25.0

markmac99, 2024-09-29:
During installation / daily refresh of the code, pip emits a warning that imreg_dft relies on an old-style egg. Eggs are deprecated and will be disabled when pip 25.0 is released (later in 2024 i think), and at that point i believe the installation will start to fail.
We therefore need to find a non-egg-based version of the module.

```
DEPRECATION: git+https://github.com/matejak/imreg_dft@master#egg=imreg_dft>'2.0.0' contains an egg fragment with a non-PEP 508 name pip 25.0 will enforce this behaviour change. A possible replacement is to use the req @ url syntax, and remove the egg fragment. Discussion can be found at https://github.com/pypa/pip/issues/11617
```

ichard26, 2025-01-12:
The easiest way to address this deprecation is to remove the version specifier from the URL fragment as it's not supported and pip has never respected it. Eggs are also deprecated, but those are going to be removed separately, and I sincerely doubt this package is using eggs (not egg-infos!) as any remotely modern tooling does not produce eggs.

markmac99, 2025-01-12:
I think you are correct, however we previously had compatability issues with other versions of imreg_dft, so we'll need to do some careful testing. 
For avoidance of doubt, i assume you mean altering requirements.txt to have 
``` bash
imreg_dft @ git+https://github.com/matejak/imreg_dft@master
```