CroatianMeteorNetwork/RMS #434: Bug in startObservationSummaryReport  - incorrectly assuming source code location

markmac99, 2024-09-29:
Richard Bassom in the UK reports a failure on the multicam installation that seems to be due to the new functionality to create a report of station status.

At line 169, the code in startObservationSummaryReport tries to import a git repo to check the RMS version. It assumes that the the folder containing the RMS config file also contains the code. On multistation builds this is incorrect and so it fails.

The code in question seems to have been introduced into RMS about 28 days ago, and may not have been noticed by multicam users as many do not reboot frequently. 
I think we need exception handling around that section of the code and/or passing in the right folder. I'll raise an issue and perhaps someone familiar with this code could take a look? 

On 29/09/2024 09:20, Richard Bassom via groups.io wrote:

I had a problem at the start of last night with my Ubuntu Linux PC RMS system running 3 cameras. I reverted to the previous RMS release to get it running again.
The RMS log shows the following, and the capture didn't start:

2024/09/28 18:23:14-INFO-Logger-line:44 - Traceback (most recent call last):
2024/09/28 18:23:14-INFO-Logger-line:44 - File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
2024/09/28 18:23:14-INFO-Logger-line:44 - return _run_code(code, main_globals, None,
2024/09/28 18:23:14-INFO-Logger-line:44 - File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
2024/09/28 18:23:14-INFO-Logger-line:44 - exec(code, run_globals)
2024/09/28 18:23:14-INFO-Logger-line:44 - File "/home/pi/vRMS/lib/python3.8/site-packages/RMS-0.1-py3.8-linux-x86_64.egg/RMS/StartCapture.py", line 1181, in <module>
2024/09/28 18:23:14-INFO-Logger-line:44 - night_archive_dir = runCapture(config, duration=duration, nodetect=cml_args.nodetect, \
2024/09/28 18:23:14-INFO-Logger-line:44 - File "/home/pi/vRMS/lib/python3.8/site-packages/RMS-0.1-py3.8-linux-x86_64.egg/RMS/StartCapture.py", line 387, in runCapture
2024/09/28 18:23:14-INFO-Logger-line:44 - log.info(startObservationSummaryReport(config, duration, force_delete=False))
2024/09/28 18:23:14-INFO-Logger-line:44 - File "/home/pi/vRMS/lib/python3.8/site-packages/RMS-0.1-py3.8-linux-x86_64.egg/RMS/Formats/ObservationSummary.py", line 169, in startObservationSummaryReport
2024/09/28 18:23:14-INFO-Logger-line:44 - repo = git.Repo(search_parent_directories=True)
2024/09/28 18:23:14-INFO-Logger-line:44 - File "/home/pi/vRMS/lib/python3.8/site-packages/git/repo/base.py", line 265, in __init__
2024/09/28 18:23:14-INFO-Logger-line:44 - raise InvalidGitRepositoryError(epath)
2024/09/28 18:23:14-INFO-Logger-line:44 - git.exc
2024/09/28 18:23:14-INFO-Logger-line:44 - .
2024/09/28 18:23:14-INFO-Logger-line:44 - InvalidGitRepositoryError
2024/09/28 18:23:14-INFO-Logger-line:44 - :
2024/09/28 18:23:14-INFO-Logger-line:44 - /home/pi/source/Stations/UK0001

g7gpr, 2024-09-29:
Working on this now. As a triage measure, I propose to remove the offending line until a proper solution is found.

Cybis320, 2024-09-29:
Make sure the XXxxxx_StartCap.desktop in the Desktop and in ~/.config/autostart is updated to:
```
[Desktop Entry]
Name=XXxxxx-Startcapture
Type=Application
Exec=lxterminal --title=XXxxxx -e "~/source/RMS/Scripts/MultiCamLinux/StartCapture.sh XXxxxx"
Hidden=false
NoDisplay=false
Icon=lxterminal
```

satmonkey, 2024-09-29:
Can we now introduce both MCP files, as discussed earlier in the email thread?
[MultiCamLinux.zip](https://github.com/user-attachments/files/17179603/MultiCamLinux.zip)
This is the gap in master for multicam now.

g7gpr, 2024-09-29:
These scripts have a couple of issues. 

Firstly

https://github.com/g7gpr/RMS/blame/74a822bc833fc18f5ac7df16c56c08e1619c80e7/Scripts/MultiCamLinux/StartCapture.sh#L44

change directory to ~/source/Stations/STATID which causes various strange things to happen; for example RMS can run a version of code which is cached somewhere, rather than the version in ~/source/RMS

Secondly the RMS_StartCapture_MCP.sh loops through all the stations starting them at 5 second intervals. If the first station to be started has not finished any pre-run work, then two station start working on the same code. This can produce strange effects.

g7gpr, 2024-09-29:
I believe this has been handled by cdec45adf64585586a66399b9336161c28956af8

dvida, 2024-09-29:
It's a quick fix that ignores the git repo part as it's not critical for observations, just for reporting.

g7gpr, 2024-09-29:
I'm working through the multicam Linux and Pi and implementations. A lot of what we are seeing needs fixing there, not in the main code.

g7gpr, 2025-03-04:
I think the main issue here is resolved - can we close this?

closed 2025-03-06 by g7gpr