CroatianMeteorNetwork/RMS #442: ..._observation_summary.txt/json

nzstevem, 2024-10-01:
Is it possible for these files to be uploaded to the archive structure (alongside the timelapse etc) each processing cycle? This would allow the Status Viewer to display them and others to collect and analysis them against each other.
Apologies if they are already and I'm just not seeing them.

g7gpr, 2024-10-01:
https://globalmeteornetwork.org/weblog//AU/AU0006/AU0006_20240930_104227_020851_detected/AU0006_20240930_104227_020851_observation_summary.json

See above for an example. Please get back if there is something strange going on.

nzstevem, 2024-10-02:
Thanks for that. I've deployed a new version of the status viewer that has a new ~ (tilde) option under each capture thumbnail which will display the camera's observation summary if one exists. If new items are added to the summary these should flow through without any code changes.

Please let me know if you have any suggestions or see any issues.

g7gpr, 2024-10-02:
There are dozens of ideas. Eg warning if lots of fits files are missing, time sync wrong. Heaps!

g7gpr, 2024-10-02:
It would be tricky, but looking to see if the commit is part of master, but more than 7 days behind head. If not on master then ignore. This is wish listing though.

nzstevem, 2024-10-02:
I'll close this out. Sounds like it might never end.

closed 2024-10-02 by nzstevem