CroatianMeteorNetwork/RMS #449: SkyFit2 does not read mp4 files using the default media_backend

satmonkey, 2024-11-05:
When opening the mp4 file by SkyFit, e.g.:
"python -m Utils.SkyFit2 /opt/RMS/FR/20241024/k/k4_241024.mp4 -c . -t 20241024_192437.0 -f 12 -n",
it ends up with the error:
----------------
Using video file: /opt/RMS/FR/20241024/k/k4_241024.mp4
Gstreamer video pipeline:
filesrc location=/opt/RMS/FR/20241024/k/k4_241024.mp4 ! matroskademux ! h264parse ! avdec_h264 ! videoconvert ! video/x-raw,format=BGR ! queue leaky=downstream max-size-buffers=100 ! appsink emit-signals=True max-buffers=100 drop=False sync=0 name=appsink
0:00:00.418337349 2546706      0x52132a0 WARN                 basesrc gstbasesrc.c:3600:gst_base_src_start_complete:<filesrc0> pad not activated yet
0:00:00.418685662 2546706      0x3ca6d20 ERROR               ebmlread ebml-read.c:139:gst_ebml_peek_id_length:<matroskademux0> Invalid EBML ID size tag (0x0) at position 0 (0x0)
0:00:00.418720199 2546706      0x3ca6d20 WARN           matroskademux matroska-demux.c:5034:gst_matroska_demux_check_parse_error:<matroskademux0> parse error, looking for next cluster
0:00:00.426618392 2546706      0x3ca6d20 WARN           matroskademux matroska-demux.c:5745:gst_matroska_demux_loop:<matroskademux0> error: got eos but no streams (yet)
-------------
However, when avoiding the gstreamer by media_backend: cv2, it reads the video file fine.

satmonkey, 2024-11-06:
As the ffmpeg seems to be more robust when dealing with different codecs, would it be possible to make the opencv:ffmpeg default choice? Even the fallback does not work here.

dvida, 2024-11-06:
I'd be happy to diagnose this, could you please provide some example data via email?

aelbazsanz, 2025-01-19:
Hello,

I have the same problem with SkyFit2. How can I change the media_backend or software settings to read the video to me?

Many thanks

satmonkey, 2025-01-20:
Now, I tried again on the master branch and a fresh pull, the same error appears.

satmonkey, 2025-01-21:
Fine - after replacing matroskademux with qtdemux, the SkyFit2 runs fine.
Is this something which can break it in other scenarios?

dvida, 2025-03-11:
I attempted a fix, it's pushed into master. Let me know if this solves it or now.

dvida, 2025-05-28:
Great, I'm closing it then!

closed 2025-05-28 by dvida