CroatianMeteorNetwork/RMS #463: Change to RMS_StartCaptiure_MCP.sh

peschman, 2024-11-15:
Recently I have seen one of the two cameras on Pi5 Bookworm fail to start properly when a cron job reboots the system in the afternoon. I believe this is due to the second camera not waiting long enough to allow the first camera to finish compiling modules refreshed by RMS_Update. Lately it has been the first camera that fails to start, because of conflict with the second camera compiling at the same time.

I suggest that RMS_StartCatpure_MCP.sh be revised to include a longer timeout, and to store the timeout value in a var so that this behavior can be echoed to the screen -- thus explaining to anyone watching that there is a delay after starting each camera. In my testing on this Pi5 increasing from 30 to 60 seconds has insured that the first camera has finished compiling before the second one begins. As a side effect, the second one starts much faster, since everything has been re-complied.

A more clever edit to the startup loop using a loop counter could only impose the longer wait interval only after the first camera start, but to keep things simple, I propose the code below for now.
Peter E.

#!/bin/bash

seconds=60
echo " Starting all configured stations post-update,"
echo "  and waiting $seconds seconds after starting each camera..."

for Dir in ~/source/Stations/*
  do
	Station=$(basename $Dir)
	echo "Starting camera ${Station}"
	lxterminal --title=${Station} -e "$HOME/source/RMS/Scripts/MultiCamLinux/StartCapture.sh ${Station}"  &
	sleep ${seconds}
  done

peschman, 2024-11-15:
I'm not able to resist changing the long delay to follow only the first camera:

#!/bin/bash

seconds=60
echo " Starting all configured stations post-update..."

loop=0
for Dir in ~/source/Stations/*
  do
	Station=$(basename $Dir)
	echo " Starting camera ${Station}"
	lxterminal --title=${Station} -e "$HOME/source/RMS/Scripts/MultiCamLinux/StartCapture.sh ${Station}"  &
	echo "  waiting $seconds seconds..."
	sleep ${seconds}
	if [[ $loop = 0 ]] ; then
	    seconds=10
	fi
	loop=1
  done
echo " All cameras started"

peschman, 2024-11-15:
Apologies for this editor stripping whitespace indents. I have tried to attach the original
[RMS_StartCapture_MCP.txt](https://github.com/user-attachments/files/17782274/RMS_StartCapture_MCP.txt)

g7gpr, 2024-11-20:
Let's have a call on Friday to get more details and make a plan.

peschman, 2025-06-07:
Denis updated the file and it has been distributed for some time now.
The issue is resolved.